cortex-lab/phy #655: stalled installation win7

amansaleem, 2015-09-28:
I am trying to install phy on Window 7, using the instructions provided. I installed POwerShell and I have the version (up to date) of C++ already installed.

The installation in the PowerShell window stops right after I start (image) 
[image]

nippoo, 2015-09-29:
How long does it stop for?

amansaleem, 2015-09-29:
I checked after 3 hours and nothing happened, so I closed the session. If
you are coming down to old street for electrophysiology club, you can have
a go yourself.

nippoo, 2015-10-06:
Pressing Ctrl+C solved this issue. Still not sure what/why happened...

closed 2015-10-06 by nippoo