cortex-lab/phy #663: correlograms sometimes don't display properly?

nsteinme, 2015-10-04:
I don't really know how to go about reproducing this, but logging the observation here: I was looking at a cluster which, in the CCG view, did not show any spikes with small ISIs. However, my registered statistic which computes these things said there should be one there. Normally I attribute this to a rounding error of some kind, where a small number of spikes in one bin on the CCG can "disappear" when the other bins are much larger. But this cluster had a very small number of spikes, so that wasn't it. I quit and re-opened phy, went back to the same cluster, and indeed I now saw a spike in a bin near zero where I expected to see one, so the plot was displaying incorrectly before I rebooted phy. Maybe related to switching the ccg view width and bin sizes, which I do a lot of?

rossant, 2016-03-26:
do you still have this bug in the latest version?

nsteinme, 2016-03-26:
Not sure, will let you know if I see it.

rossant, 2016-03-30:
OK, feel free to reopen then

closed 2016-03-30 by rossant