cortex-lab/phy #668: Cluster number confusion after 'undo'

shabnamkadir, 2015-10-14:
After moving Cluster 39 to 'good', I changed my mind and pressed 'undo'. I then decided that Cluster 39 is indeed good, but discovered that the cluster it was sending to 'good' appeared to be different. Although the GUI showed Cluster 39, the console seemingly gave random cluster numbers (55, 24):

```
22:17:53 [I] Move clusters [39] to good.
22:18:05 [I] Undo metadata_group.
22:18:14 [I] Move clusters [55] to good.
22:18:23 [I] Undo metadata_group.
22:18:40 [I] Move clusters [24] to good.
```

rossant, 2015-10-14:
Quite possible, this part of the implementation used to be overly hackish, but the next version should fix it.

rossant, 2016-03-26:
is this fixed in the latest version?

closed 2016-04-13 by rossant