cortex-lab/phy #671: spikedetekt is way too slow

nsteinme, 2015-11-03:
I even have a constructed/simulated file with literally only about 77k spikes in it, and it's taking hours to find them all. It makes no sense! There is probably some kind of really simple optimizations that can be done to make it speed up by a zillion. Whatever it is doing currently is the wrong algorithm, and it's frustrating.

rossant, 2015-11-03:
> There is probably some kind of really simple optimizations that can be done to make it speed up by a zillion

Cool, it won't take you long to do a PR then?

nsteinme, 2015-11-03:
I don't know what they are! :) I just wanted to post this issue so it is
"on the list".

rossant, 2015-11-03:
Something that comes to my mind is waveform interpolation: this is one of the bottlenecks, and it looks like [it is done on all channels currently](https://github.com/kwikteam/phy/blob/master/phy/traces/waveform.py#L150), whereas it should probably be done on the unmasked channels only, especially with huge probes likes yours...

rossant, 2016-03-26:
closing since spikedetekt has been moved into [klusta](https://github.com/kwikteam/klusta), feel free to reopen there if necessary

closed 2016-03-26 by rossant