cortex-lab/phy #674: float64 data default scaling / max zoom issues

nippoo, 2015-12-01:
Need to test with float64 data (I've got some real ones here), it takes about a minute of constantly zooming in to get to the correct zoom ;) and the FV is limited to a maximum zoom which is too small.

rossant, 2015-12-02:
what branch? what view? what is the distribution of the data? which arrays are float64 exactly?

The new branch contains a specific mechanism for data normalization (based on percentiles rather than absolute max) based on CPU transforms. We need to ensure that conversion to float32 only happens at the very last moment, right before GPU uploading.

nippoo, 2015-12-02:
master, FeatureView/WaveformView/TraceView.

.dat is in float64. It's not actually a bug, it's just an internal note that we should have some datasets of different dtypes in the automated testing.

closed 2019-06-13 by rossant