cortex-lab/phy #677: New SpikeDetekt doesn't give same results as old SpikeDetect

nippoo, 2015-12-15:
Using the same (default) parameters, new SpikeDetekt (phy 0.3.0) returns slightly more spikes, and differently-timed spikes to phy 0.2.0.

[new-old-npy.zip](https://github.com/kwikteam/phy/files/62700/new-old-npy.zip)

nippoo, 2015-12-15:
(Attached file is using hybrid-10sec.dat)

rossant, 2015-12-15:
are the new results sensible or nonsensical?
have you tried with a single chunk? overlapping is handled differently (using dask), that might be one explanation

nippoo, 2015-12-15:
New results look sensible at first glance (haven't investigated the differences too closely yet) but there are differences around 1000 samples into the dataset, so << 1 second (the chunk size)

closed 2016-03-24 by rossant