cortex-lab/phy #690: installation instructions in README don't seem to work

neuromusic, 2016-04-10:
```
justin@niao:~/Code/ephys-analysis$ conda env create -n phy2 -f ~/Downloads/environment.yml 
Fetching package metadata: ....Error: HTTPError: 401 Client Error: UNAUTHORIZED: https://conda.binstar.org/kwikteam/noarch/
```

neuromusic, 2016-04-10:
after a bit of sleuthing, it appears that a binstar (anaconda cloud) login must be setup and active (by calling `binstar login` or `anaconda login`) before these instructions will work

rossant, 2016-04-11:
Very weird. This is the first time I see this. @nippoo could you look into it? Thanks

nippoo, 2016-04-11:
Why's it trying to fetch noarch? What system / CPU do you have?

neuromusic, 2016-04-11:
```
Welcome to Ubuntu 14.04.4 LTS (GNU/Linux 3.13.0-85-generic x86_64)

 * Documentation:  https://help.ubuntu.com/

Last login: Fri Apr  1 22:22:03 2016 from 209.134.92.11
justin@machine:~$ lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                8
On-line CPU(s) list:   0-7
Thread(s) per core:    2
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 60
Stepping:              3
CPU MHz:               800.000
BogoMIPS:              7000.04
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              8192K
NUMA node0 CPU(s):     0-7
```

neuromusic, 2016-04-11:
also, the output above was with conda 3.18. I updated to 4.0.5 and the error was slightly different (but I can't recall). trying to figure out how to logout of binstar or clear the login info to recreate it.

neuromusic, 2016-04-11:
hmm. ok it looks like I'm an anon user again

```
justin@machine:~$ binstar whoami
Using binstar api site https://api.binstar.org
Anonymous User
```

and it's downloading just fine.

I can't recreate the error on an updated conda, so I'm just going to close the issue

carry on. excited to test out the Kwik-GUI

closed 2016-04-11 by neuromusic