cortex-lab/phy #691: Cluster_group 1 saved as its label 'Mua' in kwik files when using kwik-gui

ablot, 2016-04-12:
When I use phy kwik-gui, the channel groups are stored as 0, 'Mua', 2, 3 (I look in the .kwik, /channel_groups/0/clusters/main/XX/cluster_group). 
I'm not sure if it is a bug or not (and if I should report it in phycontrib, phy or klusta). Is the mixture of group name and group ID expected? I think it would be a bit easier to have only integers or only group names.

rossant, 2016-04-12:
Thanks for the report. Yes, it is a bug in the klusta package (which is the package that implements the Kwik format). I just pushed a fix on the master branch of klusta. Now, integers should always be used in the Kwik file, whereas strings are used in the `KwikModel`. Feel free to try it and close the issue if it works. You might need to re-open and re-save your file after you updated klusta.

ablot, 2016-04-12:
I works but I cannot reopen the files created with the previous klusta version without [a small change](https://github.com/ablot/klusta/commit/f0d2c762e5fad4376a2bdddb7373eb9d7cc7764e)

I haven't made a PR, since I assume that you don't really fancy that kind of ad hoc dirty check :)

rossant, 2016-04-12:
OK, I pushed a proper fix, please try again :)

ablot, 2016-04-12:
Perfect! It seems to work nicely

closed 2016-04-12 by ablot