cortex-lab/phy #694: Lasso tool missing

ananmoran, 2016-04-17:
Hi,
It might that I've failed to locate it, but it seems that the lasso tool in the gui was removed in the new version of phy-gui. Any intentions to put it back in again? I found the ability to correct the automatic clustering essential.

Thanks
Anan

rossant, 2016-04-17:
Try ctrl + click on the polygon points

ananmoran, 2016-04-17:
Works. Thanks. I did not find the documentation for that functionality. 
Another desired functionality of the polygon might be to remove undesired spikes from a selected cluster (for example by pressing ctrl+shift). would you consider adding that functionality? (or is it already implemented?)

rossant, 2016-04-17:
Yes I need to add this to the doc!
If you select some spikes and split with `k`, you get two new clusters, one with the selected spikes, another with the rest of the spikes. Then you can move the undesired cluster to the "noise" group. Does that answer your question or didn't I understand correctly?

ananmoran, 2016-04-17:
Exactly! That's perfect. 
Much obliged

rossant, 2016-04-17:
you're welcome!

closed 2016-04-17 by rossant