cortex-lab/phy #695: Error when moving cluster after merging f66848e

ablot, 2016-04-17:
Since I re-based my fork, I get an error when I try to move a cluster. It seems to come from f66848e

```
17:14:07 [E] cli:37               An error has occurred (TypeError): <functools.partial object at 0x7fc219c6b5d0> is not a Python function
17:14:07 [D] cli:40               Traceback (most recent call last):
  File "/home/blota/Documents/PyCharmProjects/phy/phy/gui/actions.py", line 70, in wrapped
    argspec = inspect.getargspec(f)
  File "/home/blota/anaconda3/envs/klusta/lib/python2.7/inspect.py", line 816, in getargspec
    raise TypeError('{!r} is not a Python function'.format(func))
TypeError: <functools.partial object at 0x7fc219c6b5d0> is not a Python function
```

ablot, 2016-04-17:
#696  seems to fix the problem. But since I don't know why partial functions are passed to `wrapped` with no `args`, I'm not sure it is the appropriate fix at all.

crnolan, 2016-05-19:
I ran across the same problem but didn't see the issue filed, or the associated fix. I've created #698 to add the handling for situations with different combinations of defaults and arguments supplied to the partial.

PR #698 (2016-07-20): Deal with partial functions when used as wrapped callbacks.
Update to #696, I didn't know how to merge with the existing pull request. This has the same modification, but with handling of the defaults and arguments supplied to the partial (#696 works when the arguments supplied to the partial also have defaults).

Fixes #695
closed 2016-06-28 by ablot