cortex-lab/phy #715: phy featureView

llobetv, 2016-10-17:
Hi,

To prevent it's my first come on github, 

I wanna propose a modification for featureView of phy, : 
[image]
to
[image]

where 3 features are plot and legend works (more readable).

modification code  from phy/cluster/manual/views.py :

def _dimensions_matrix(channels, n_cols=None, top_left_attribute=None):
      """
      y0,x0   y0,x1   y0,x2   time,y0 
      y1,x0   y1,x1   y1,x2   time,y1
      y2,x0   y2,x1   y2,x2   time,y2
      time,x0   time,x1   time,x2   time,time
      """
     #here i change matrix to get 3 features wich is default parameter of klusta and keep same parameter for 1 line/1 column to make it more readable, could be good to easily change number of feature plot?

    # Generate the dimensions matrix from the docstring.
    ds = inspect.getdoc(_dimensions_matrix).strip()
    x, y = channels[:2]

    def _get_dim(d):
        if d == 'time':
            return d
        assert re.match(r'[xy][012]', d) **# change [01] to [012] to plot third PCA**
        c = x if d[0] == 'x' else y
        f = int(d[1])
        return c, f
    dims = [[_.split(',') for _ in re.split(r' +', line.strip())]
            for line in ds.splitlines()]
    x_dim = {(i, j): _get_dim(dims[i][j][0])
             for i, j in product(range(4), range(4))}
    y_dim = {(i, j): _get_dim(dims[i][j][1])
             for i, j in product(range(4), range(4))}
    return x_dim, y_dim 

and :

    def _plot_features(self, i, j, x_dim, y_dim, x, y,
                       masks=None, clu_idx=None):
        """Plot the features in a subplot."""
        assert x.shape == y.shape
        n_spikes = x.shape[0]

        if clu_idx is not None:
            color = tuple(_colormap(clu_idx)) + (.5,)
        else:
            color = (1., 1., 1., .5)
        assert len(color) == 4

        # Find the masks for the given subplot channel.
        if isinstance(x_dim[i, j], tuple):
            ch, fet = x_dim[i, j]
            # NOTE: we add the cluster relative index for the computation
            # of the depth on the GPU.
            m = masks[:, ch] * .999 + (clu_idx or 0)
        else:
            m = np.ones(n_spikes) * .999 + (clu_idx or 0)

        # Marker size, smaller for background features.
        size = self._default_marker_size if clu_idx is not None else 1.

        self[i, j].scatter(x=x, y=y,
                           color=color,
                           masks=m,
                           size=size,
                           data_bounds=None,
                           uniform=True,
                           )

	
        if i == 0:         
            # HACK: call this when i=0 (first line) but plot the text
            # in the last subplot line. This is because we skip i > j
            # in the subplot loop.
            i0 = (self.n_cols - 1)	   
            dim = x_dim[j, i]                          ###old : dim = x_dim[i0, j] #if j < (self.n_cols - 1) else x_dim[i0, 0],
            self[i0, j].text(pos=[0., -1.], 
                             text=str(dim),
                             anchor=[0., -1.04],
                             data_bounds=None,
                             )
        if j == (self.n_cols - 1):                    ###instead of j=0 to take last column coordinate for time
            self[i, 0].text(pos=[-1., 0.],
                            text=str(y_dim[j, i]),    ### instead of  text=str(y_dim[i, j]), seems to be an error of line/column, => j column fix and line which varies**
                            anchor=[-1.03, 0.],
                            data_bounds=None,
                            )

and i remove :                 
    # Skip lower-diagonal subplots.
    #if i > j:  
             #    continue
**from **    def on_select(self, cluster_ids=None): to plot all graph.

llobetv, 2016-11-07:
My proposal doesn't seems to please you :p, maybe can i make something that generate plot matrix depend of how many feature you want on the same idea, and let people choose it from menu bar?

rossant, 2016-11-07:
sorry I didn't have time to review this... :( I'll try to have a look later this week
@nsteinme do you want to comment on this proposition?

nsteinme, 2016-11-07:
We talked a lot about this and settled on the current view for phy. So I am opposed to changing the default. But I think it would be great if the feature grid were customizable in a sensible way, since it's evident that people have various use cases for it. I don't know if it's best for there to be some kind of cycle-through features with a hotkey, or a way to specify the layout you want in a user_settings file? Don't know.

rossant, 2016-11-07:
@nsteinme good idea, I can make it such that users can customize the dimensions. I'll change the title of this issue accordingly.
@llobetv note that the labels in the feature view have been fixed recently by @mspacek

mspacek, 2016-11-07:
The proposed layout does seem like a better use of space, and it can't hurt to see an extra PC on each channel, as well as having time explicitly labelled. An arbitrary layout in a Python syntax user settings file might be nice, but I guess such a file doesn't exist for phy like it does for, say, klusta (unless there's something lurking in the .phy folder). If there aren't any other proposals for different FeatureView layouts, then perhaps a key to toggle between the default and the above proposed new layout would be perfectly adequate.

mspacek, 2016-11-08:
I just realized that with the above layout proposal, you wouldn't be able to compare PCs within a channel. So there's obviously a tradeoff. As the number of dimensions increases, the number of pairwise comparisons increases with the square.

rossant, 2016-11-08:
There is a config and plugin system in phy that would make this possible. It's not ideal and the system might change a bit in the near future, but it could be a start.

llobetv, 2016-11-08:
Yeah to compare PC on one channel you have to select twice time the same channel which is not ideal if you want to have quick view of pple channel, but the impossibility to see third component  (for klusta which works with 3 feature by default) or more is really problematic for manually change in clusters. There is no easy way to add a choose of kind of feature or number of plot feature on menu bar?

Nice info about featureview label, i was looking to fix it

rossant, 2016-11-10:
@llobetv not at the moment, I'll look into it

rossant, 2019-06-13:
done in the plugin examples, in the dev branch

closed 2019-06-13 by rossant