cortex-lab/phy #717: Make TraceView use vertical spatial channel order

mspacek, 2016-10-18:
Unless I'm doing something wrong with my custom `.prb` file, it seems that the TraceView channels are sorted in numerical order. It would be much better to sort them in vertical spatial order, so that spikes show up as spatially localized events. Depending on the channel mapping, the waveforms of a multichannel spike can end up scattered across the probe. Vertical spatial order would also make the TraceView more directly comparable to the WaveformView.

rossant, 2016-10-22:
I think it could be a good idea. @nsteinme @yger ?

nsteinme, 2016-10-22:
I definitely think it's best to sort them in vertical spatial order - as @mspacek says, that makes it comparable to the waveform view, and matches your logical conception of the relationship between channels. I thought that's how it already was! Mine come out that way - but for my recent recordings, the two are identical so I guess I just didn't notice it was doing something different.

yger, 2016-10-23:
Indeed, I also think this has to be the case. It would be easier to read for vertical probes layout. For square MEA/probes, this will still be annoying, but I do not think there is much you can do about that.

closed 2016-10-27 by rossant

mspacek, 2016-10-27:
Thanks Cyrille, but from what I can tell, the TraceView is not in vertical spatial order. Here's a screenshot:

[image]

The channel labels in the TraceView don't match vertically with those in the WaveformView. Also, each spike in the TraceView is scattered across the probe, instead of being spatially localized. Here are my crresponding .prm and .prb files:

[prm_prb_files.zip](https://github.com/kwikteam/phy/files/556113/prm_prb_files.zip)

rossant, 2016-10-27:
could you send me the full dataset?

mspacek, 2016-10-27:
Sure. Here's a 10 sec excerpt of data from a single column A1x32 NeuroNexus probe. All the relevant files should be there. Let me know once you've downloaded it, so I can delete it from this file sharing site:

https://mega.nz/#!tRtD1LTS!mG6cGNiKv-m-P9yAK4o6frNgUDJWFKWWuB6WMlYkdik

rossant, 2016-10-27:
Thanks. Downloaded, I'll have a look.

rossant, 2016-10-27:
OK I pushed a fix on master, can you try again?

micheleacox, 2016-10-27:
For clarification, to pull fixes from the master, do we use: 

pip install phy phycontrib --upgrade

rossant, 2016-10-27:
No, this will upgrade to the latest release. Use the following command to upgrade to the development version on master:

```
pip install git+https://github.com/kwikteam/phy.git git+https://github.com/kwikteam/phycontrib.git --upgrade
```

mspacek, 2016-10-28:
Yup, that did it. Thanks! Ha, having to argsort an argsort sounds familiar.

PR #719 (2016-10-27): Display traces in vertical channel order in trace view
* Closes #716.
* Closes #717.