cortex-lab/phy #728: Problem with ACG and feature template view

DCampa, 2016-11-18:
Hello,
I am using kilosort+phy to spikesort data from 384 channels recorded with a neuropixel O3 probe. 

ACG ISSUE:
 I found something unclear with the ACG. In the figure below (left panel) you can see in phy what it seems a quite well isolated unit from the waveform view (cluster 799). However, looking at the refractory period in the ACG in looks like this cluster should actually be classified as multiunit (no refractory period present). 

[image]

I have then loaded in Matlab the spike times of that cluster and, if you look at the histogram of the inter spike interval, what you see is that you have 15 events only in which the ISI is between 1ms and 2ms and ~50 events only in which it is between 2 and 3ms (figure below, right panel). I do not understand how this can be consistent with the ACG, also considering that the total number of spikes in the cluster is 2768. I would have expected to see a much clearer refractory period in the ACG. Could you please help me with this issue?

[image]

FEATURE TEMPLATE VIEW ISSUE:
I have a separate issue with the feature template view. Basically when I select a cluster it looks empty or with only one dot. Is it possible to fix this?

Thank you for your help,
Dario

marius10p, 2016-11-18:
Hi Dario,

Have you tried changing the binning of the ACG? It might be too zoomed out for you to see the bins near zero. Also, keep in mind the waveform view only shows you a small subset of the spikes (for display purposes). Also, that looks like a narrow-spiking unit, so it is not impossible for it find two spikes within 1ms of each other. Check the raw data if you want to convince yourself. 

We are aware of the feature template view issue for 384 channels, at least in some cases, and are working to fix it. Will let you know here when that is fixed. 

Best,
   Marius

DCampa, 2016-11-18:
Hi Marius,

Thank you for your quick reply. I have already tried to zoom in, but the gap you see in the middle of the ACG correspond to 0 ms only. What make me a bit suspicious is actually looking at the raw data together with the cluster's spike times. In fact it happens only once that two events are closer than 1 ms, and only 45 times closer than 2 ms (as you can see already from the histogram above). So I do not understand how in the ACG I see those 2 large peaks at +-1ms. 
Thank you, 
Dario

marius10p, 2016-11-18:
Maybe @rossant has a better idea for what's happening with the ACG view here?

rossant, 2016-11-18:
Have you tried to change the bin size in the CorrelogramView menu?

Have you tried to compute and plot the CCG in Matlab to compare it with phy?

DCampa, 2016-11-20:
Thanks for the replies. 

I have tried to zoom in and out but this does not seem to solve the issue.
Set_bin = 1 ms; Set_window = 50 ms

[image]

Set_bin = 1 ms; Set_window = 10ms

[image]

About computing the ACG in matlab I have done it in the following way (idx is the vector of the spike times in second for the events in the cluster):

  _figure
 xx = zeros(1,round(idx(end)*30000));
 xx(round(idx*30000)) = 1;
  [one two] = xcorr(xx,300);
  one(301) = 0
  figure,plot(two, one,'*')
  xlabel('time (sapling rate 30000Hz)')_

This is the ACG:

[image]

In the x axis of the ACG, 1 ms corresponds to 30. So you can see that the cluster shows a refractory period (~0 correlation from -26 to +26), but this is not visible in the ACG from phy.

Please let me know if my explanation is unclear or you need further information.

Thank you very much for your help.

rossant, 2016-11-21:
can you try set bin=0.1 ms?

DCampa, 2016-11-22:
This is the ACG with set bin=0.1 ms
[image]

rossant, 2016-11-22:
So is there a problem? There is a refractory period here, no?

DCampa, 2016-11-22:
I think the 0.1ms one makes more sense. But it is still unclear to me why the  bar a time = 1 ms of the ACG  with set bin = 1ms is so high, and higher than the bar at time = 2 ms. Given the refractory period visible in the 0.1 ms ACG for this unit, should not be the opposite?
Thank you

rossant, 2016-11-22:
I am not sure I understand the problem... @nsteinme might have a better idea?

closed 2016-12-12 by rossant