cortex-lab/phy #729: Problem with trace view in TemplateGUI

nicklesica, 2016-11-29:
It seems like all of the clusters are always colored in the trace view no matter which I select. My understanding is that only the spikes in the selected cluster should be shown.

There are also two lines displayed in trace view for each channel. Is that expected?

Thanks,

Nick Lesica

rossant, 2016-11-29:
This is the intended behavior at the moment. Use Alt+Left/Right to force highlighting the spikes of the selected clusters (this will be improved soon).

For each channel, you see the raw traces, and raw traces minus templates.

nicklesica, 2016-11-29:
Thanks for the quick reply!

Sorry, perhaps I am being thick, but Alt+Left/Right seems to just shift the traces to the left or right, not change anything about the highlighting.

rossant, 2016-11-29:
Yes, if the shown interval happens to contain spikes from selected clusters, they will appear highlighted, otherwise nothing will change. An upcoming update will automatically jump to a time window that contains spikes from the selected clusters.

nicklesica, 2016-11-29:
Sorry, I'm still not getting it. In the screenshot below, I've selected cluster 21, but it seems that all spikes from all clusters are highlighted in the trace view. Alt+Left/Right has no effect on the highlighting, but simply slides the trace view to the left or right. Is there are way that I can see a trace view with only the spikes from cluster 21 highlighted?

[image]

rossant, 2016-11-29:
Ah, well in the screenshot spikes from the selected clusters are highlighted in a brighter color than other spikes, but all spikes are highlighted. There's no other display option at the moment although we could implement one if necessary.

nicklesica, 2016-11-29:
Oh, I see. OK, that is very hard for me to see, even now that I know to look for it. Perhaps it is because I am colorblind? Yes, I think if you could add a way to toggle off the colors for all of the non-selected clusters in a future version that would help (at least me) a lot. 

Thanks!

rossant, 2016-11-29:
Even for a non-colorblind it is definitely suboptimal at the moment. I'll improve this.
In the meantime @nsteinme let me know if you have any other comments that would be helpful to make this view clearer and easier to read.

nsteinme, 2016-11-29:
I think the option to toggle such that every spike is uncolored except the selected one(s) is a great idea. The other really important feature for this view that's not currently implemented is the ability to jump to places in the traces that actually contain the spikes of the selected unit.

rossant, 2016-11-29:
Both options are now implemented in the ongoing pull request, to be merged within the next few weeks.

nicklesica, 2016-11-30:
Great, thanks!

closed 2016-11-30 by nicklesica