cortex-lab/phy #730: More functionality for feature template view

nicklesica, 2016-12-02:
Hi, 

A couple of suggestions:

1) Make it possible to cut clusters in the feature template view in the same way as in feature view. I've found many examples where it is clear from feature template view that a cluster should be split, but none of the projections in feature view are good for doing it.

2) When only one cluster is selected, feature template view could show the 1-D histogram of template amplitudes for that cluster. If that histogram is sharply truncated at low values, that is a good indication that either the threshold should be lowered or that the cluster is not really separable from the background. This information can be obtained from the 2-D plots, but it would be nice not to have to choose the second cluster just for that purpose. 

Thanks,

lshaheen, 2016-12-02:
Hi Nick,
 I'm not a part of the development team, but have been making plugins for use in our lab and have made the amplitude plugin you mentioned. I've been meaning to do a pull request to get it into phy-contrib but haven't yet. In the meantime you can find it [here](https://github.com/LBHB/phy-contrib/blob/master/phycontrib/LBHB_plugins/AmplitudeHistogram.py).

nicklesica, 2016-12-05:
Thanks. I've downloaded your .py file, but I'm not sure what to do with it to add its functionality to Phy ...

lshaheen, 2016-12-05:
Check out the comments at the top of the file for instructions. First, put it in your plugins folder ( should be ~/.phy/plugins/). Then add this line to your config file (should be ~/.phy/phy_config.py):

c.TemplateGUI.plugins = ['AmplitudeHistogram']

If you're using the kwik-gui you might be able to change TemplateGUI to KwikGUI to get it to work, but I haven't tested it.

nicklesica, 2016-12-06:
Thanks. 

I'm sorry, but I still don't get it. 

I'm using windows, not linux. 

I don't seem to have a .phy or a plugins directory.

When I search my drive for phy_config.py, nothing comes up. 

Do you have any idea where the relevant folders/files might be on Windows?

Thanks again for your help,

nicklesica, 2016-12-06:
Sorry, I found the right place now. 

Thanks again.

nicklesica, 2016-12-06:
And it works, thanks!

lshaheen, 2016-12-06:
Great! Where are the folders in Windows?

nicklesica, 2016-12-07:
In the analogous place: the user's home directory on the drive where the OS is installed. 

I was looking on a different drive where the actual Phy package is.

And there doesn't seem to be a plugins directory by default, so I created one.

Thanks again.

rswilliamson, 2017-03-30:
I've actually been trying to get some of these plugins working on my machine (running Windows10), but am having trouble getting them to work. Can anyone help me debug this phy --debug (it seems like the plugins are not being attached after the phy_config file has been read:

(phy) J:\Sensory_Characterization\CCol004\RSW013017\Sorted_Data\Penetration_1>phy template-gui params.py --debug
18:03:44 [D] config:46            Load config file `C:\Users\williar\.phy\phy_config.py`.
18:03:45 [D] plugin:35            Register plugin `phycontrib.kwik_gui.gui.KwikGUIPlugin`.
18:03:45 [D] plugin:35            Register plugin `phycontrib.template.gui.TemplateGUIPlugin`.
18:03:45 [D] plugin:35            Register plugin `phycontrib.utils.precache.PrecachePlugin`.
18:03:45 [D] plugin:35            Register plugin `phycontrib.utils.saveprompt.SavePromptPlugin`.
18:03:45 [D] plugin:72            Scanning `C:\Users\williar\.phy\plugins`.
18:03:45 [D] plugin:77            Found plugin module `AmplitudeHistogram.py`.
18:03:45 [D] plugin:35            Register plugin `AmplitudeHistogram.AmplitudeHistogram`.
18:03:45 [D] plugin:77            Found plugin module `SpikeSNR.py`.
18:03:45 [D] plugin:35            Register plugin `SpikeSNR.SpikeSNR`.
18:03:45 [D] cli:107              Attach plugin `phycontrib.kwik_gui.gui.KwikGUIPlugin` to CLI.
18:03:45 [D] cli:107              Attach plugin `phycontrib.template.gui.TemplateGUIPlugin` to CLI.
18:03:45 [D] gui:253              Loading traces at `data.dat`.
18:03:45 [D] gui:268              Loading amplitudes.
18:03:45 [D] gui:277              Loading 1113491 spike clusters.
18:03:45 [D] gui:283              Loading spike templates.
18:03:45 [D] gui:289              Loading spike samples.
18:03:45 [D] gui:293              Loading templates.
18:03:45 [D] gui:299              Loading the whitening matrix.
18:03:45 [D] gui:303              Loading unwhitened templates.
18:03:45 [D] gui:321              Loading similar templates.
18:03:45 [D] gui:326              Loading channel mapping.
18:03:45 [D] gui:334              Loading channel positions.
18:03:45 [D] gui:339              Loading features.
18:03:45 [D] gui:370              Loading template features.
18:03:45 [D] gui:424              HP filtering the data for waveforms
18:03:45 [D] gui:453              Loading the cluster groups.
18:03:45 [D] context:56           Initialize joblib cache dir at `J:\Sensory_Characterization\CCol004\RSW013017\Sorted_Data\Penetration_1\.phy`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_masks`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_features`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_waveforms`.
18:03:45 [D] context:80           Load memcache for `phycontrib.template.gui.get_waveforms_amplitude`.
18:03:45 [D] context:80           Load memcache for `phycontrib.template.gui.get_waveform_lims`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_feature_lim`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_trace_lim`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_close_clusters`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_probe_depth`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_correlograms`.
18:03:45 [D] context:80           Load memcache for `phy.cluster.manual.controller.spikes_per_cluster`.
18:03:45 [D] context:80           Load memcache for `phycontrib.template.gui.get_amplitudes`.
18:03:45 [D] context:80           Load memcache for `phycontrib.template.gui.get_cluster_templates`.
18:03:45 [D] context:80           Load memcache for `phycontrib.template.gui.get_cluster_pair_features`.
18:03:45 [D] context:80           Load memcache for `phycontrib.template.gui._cluster_template_similarities`.
18:03:45 [D] context:80           Load memcache for `phycontrib.template.gui._sim_ij`.
18:03:45 [D] context:128          The file `J:\Sensory_Characterization\CCol004\RSW013017\Sorted_Data\Penetration_1\.phy\new_cluster_id.json` doesn't exist.
18:03:45 [D] config:46            Load config file `C:\Users\williar\.phy\phy_config.py`.
18:03:45 [D] gui:428              Load the GUI state from `C:\Users\williar\.phy\TemplateGUI\state.json`.
18:03:47 [D] utils:51             Get box size for 32 points.
18:03:47 [D] gui:171              Load the geometry state.
18:03:54 [D] gui:177              Save the geometry state.
18:03:54 [D] gui:433              Save the GUI state to `C:\Users\williar\.phy\TemplateGUI\state.json`.
18:03:54 [D] gui:433              Save the GUI state to `C:\Users\williar\.phy\TemplateGUI\state.json`.
18:03:54 [D] gui:433              Save the GUI state to `C:\Users\williar\.phy\TemplateGUI\state.json`.
18:03:54 [D] gui:433              Save the GUI state to `C:\Users\williar\.phy\TemplateGUI\state.json`.
18:03:54 [D] gui:433              Save the GUI state to `C:\Users\williar\.phy\TemplateGUI\state.json`.
18:03:54 [D] gui:433              Save the GUI state to `C:\Users\williar\.phy\TemplateGUI\state.json`.
18:03:54 [D] gui:433              Save the GUI state to `C:\Users\williar\.phy\TemplateGUI\state.json`.
18:03:54 [D] gui:433              Save the GUI state to `C:\Users\williar\.phy\TemplateGUI\state.json`.
[... cut ...]
18:03:54 [D] context:91           Save memcache for `phy.cluster.manual.controller.get_correlograms`.
18:03:54 [D] context:91           Save memcache for `phycontrib.template.gui.get_waveform_lims`.
18:03:54 [D] context:91           Save memcache for `phycontrib.template.gui.get_amplitudes`.
18:03:54 [D] context:91           Save memcache for `phy.cluster.manual.controller.get_mean_waveforms`.
18:03:54 [D] context:91           Save memcache for `phycontrib.template.gui._sim_ij`.
18:03:54 [D] context:91           Save memcache for `phy.cluster.manual.controller.get_probe_depth`.

rossant, 2017-03-31:
Have you put `c.TemplateGUI.plugins = ['AmplitudeHistogram']` in `C:\Users\williar\.phy\phy_config.py`?

rswilliamson, 2017-03-31:
Yes (and both plugins are in the correct directory). My phy_config.py file looks like:

# You can also put your plugins in ~/.phy/plugins/.

from phy import IPlugin

try:
    import phycontrib
except:
    pass

# Plugin example:
#
# class MyPlugin(IPlugin):
#     def attach_to_cli(self, cli):
#         # you can create phy subcommands here with click
#         pass

c = get_config()
c.Plugins.dirs = [r'C:\Users\williar\.phy\plugins/']
c.TemplateGUI.plugins = ['SpikeSNR','AmplitudeHistogram']

rswilliamson, 2017-03-31:
So, I gather from some digging that it might be because it's only the development version that allows plugins? If so, I'm now having trouble getting the development version to work. I've followed the instructions here https://github.com/kwikteam/phy to the letter - and have just installed the development instructions as per those instructions. However, now when I try to hit
phy template-gui params.py

I get an error:
(phy) J:\Sensory_Characterization\CCol004\RSW013017\Sorted_Data\Penetration_1>phy template-gui params.py
Usage: phy [OPTIONS] COMMAND [ARGS]...

Error: No such command "template-gui".

Any ideas?

Sarruedi, 2017-07-20:
I tried exactly the same steps as described above to use the AmplitudeHistogram view but always get an error which states that the plugin could not be found. I also had no plugins folder to begin with an created in and stored the AmplitudeHistogram.py file in it. I also update the config file. I currently do not run the development version as I had trouble installing it. Do you have any suggestions on how to get the plugins to run?

dianamaro, 2017-10-09:
I can see the AmplitudeHistogram view in the KwikGui but it does not work and shows the following error: 'KwikController' object has no attribute '_get_amplitudes'. Does anyone know how to change the code for it to work in the KwikGui?

rossant, 2019-06-13:
some of it done in dev. Open a new issue for a new suggestion

closed 2019-06-13 by rossant