cortex-lab/phy #731: I don't seem to have a cluster group file

nicklesica, 2016-12-07:
I don't seem to have cluster_groups.csv

I'm using rez2Phy to write the Phy files from the output of Kilosort, but that function doesn't seem to write the cluster_groups.csv file. 

When I change the group of a cluster in Phy it seems to work and the terminal says:

12:48:29 [I] gui_component:219    Change metadata_group for clusters 0 to noise.

But then if I exit Phy and start it again, the change in group is not remembered.

Any advice appreciated, thanks!

rossant, 2016-12-07:
Have you done ctrl+s to save? 
I should probably enable the saving reminder by default...

rossant, 2016-12-07:
Also, we could have an option where saving is automatic after each action (à la gmail, google docs, etc.), but I don't know if that would be desirable.

nicklesica, 2016-12-07:
Oh, great, thanks, that worked. 

I think autosave is probably a bad idea. But maybe put a 'Save' entry in the 'File' menu. I looked there and when I didn't see one I figured saving was happening automatically. 

Thanks!

closed 2016-12-07 by nicklesica