cortex-lab/phy #734: kwik gui not showing features or correlograms

elor-arieli, 2017-01-01:
when i try and open a kwik file with kwik gui it shows me the waveforms (although with an offset) and doesnt show correlograms or features.

i get this error in the CMD: OSError: "exception: access violation reading 0x0000000000000008"
and then this:

"ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> for DrawEvent
15:13:02 [E] logs:326             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> for DrawEvent
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 2
15:13:03 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 2
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 2
15:13:03 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 2
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 3
15:13:04 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 3
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 3
15:13:04 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 3
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 4
15:13:05 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 4
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 4
15:13:05 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 4
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 5
15:13:05 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 5
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 5
15:13:05 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 5
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 6
15:13:06 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 6
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 6
15:13:06 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 6
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 7
15:13:07 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 7
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 7
15:13:07 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 7
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 8
15:13:08 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 8
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 8
15:13:08 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 8
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 9
15:13:09 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 9
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 9
15:13:09 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 9
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 10
15:13:09 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 10
ERROR: Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 10
15:13:10 [E] logs:332             Invoking <bound method FeatureView.on_draw of <FeatureView (PyQt4) at 0xc838a90>> repeat 10
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 20
15:13:24 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 20
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 30
15:13:24 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 30
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 40
15:13:31 [E] logs:332             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> repeat 40"

elor-arieli, 2017-01-01:
ERROR: Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> for DrawEvent
15:13:02 [E] logs:326             Invoking <bound method CorrelogramView.on_draw of <CorrelogramView (PyQt4) at 0xc7f7518>> for DrawEvent
WARNING: Traceback (most recent call last):
  File "d:\users\ananm4\miniconda3\envs\phy\lib\runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\Users\AnanM4\Miniconda3\envs\phy\Scripts\phy.exe\__main__.py", line 9, in <module>
    sys.exit(phy())
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\click\core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\click\core.py", line 696, in main
    rv = self.invoke(ctx)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\click\core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\click\core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\click\core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "D:\Users\AnanM4\.phy\plugins\gui.py", line 155, in cluster_manual
    ctx, globals(), locals())
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\phy\utils\cli.py", line 76, in _run_cmd
    return exec_(cmd, glob, loc)
  File "<string>", line 1, in <module>
  File "D:\Users\AnanM4\.phy\plugins\gui.py", line 131, in _run
    run_app()
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\phy\gui\qt.py", line 106, in wrapped
    return func(*args, **kwargs)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\phy\gui\qt.py", line 127, in run_app
    return QT_APP.exit(QT_APP.exec_())
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\app\backends\_qt.py", line 631, in paintGL
    self._vispy_canvas.events.draw(region=None)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\util\event.py", line 436, in __call__
    self._invoke_callback(cb, event)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\util\event.py", line 452, in _invoke_callback
    self, cb_event=(cb, event))
  << caught exception here: >>
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\util\event.py", line 448, in _invoke_callback
    cb(event)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\phy\plot\base.py", line 300, in on_draw
    visual.on_draw()
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\phy\plot\base.py", line 73, in on_draw
    self.program.draw(self.gl_primitive_type)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\gloo\program.py", line 467, in draw
    canvas.context.flush_commands()
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\gloo\context.py", line 164, in flush_commands
    self.glir.flush(self.shared.parser)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\gloo\glir.py", line 158, in flush
    parser.parse(self._filter(self.clear(), parser))
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\gloo\glir.py", line 401, in parse
    self._parse(command)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\gloo\glir.py", line 379, in _parse
    ob.set_shaders(*args)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\gloo\glir.py", line 568, in set_shaders
    gl.glLinkProgram(self._handle)
  File "d:\users\ananm4\miniconda3\envs\phy\lib\site-packages\vispy\gloo\gl\_gl2.py", line 957, in glLinkProgram
    nativefunc(program)

rossant, 2017-01-01:
It might be an OpenGL problem, could you make sure that your graphics drivers are up to date?
What's the spec of your computer?

elor-arieli, 2017-01-02:
i have an i7-6700k, 16GB of ddr4 ram, 256gb ssd and 1TB hdd, intel graphics 530 (fully updated driver)

rossant, 2017-01-02:
OK, could you install [OpenGL extension viewer](http://www.realtech-vr.com/glview/) and tell me what OpenGL and GLSL version are supported? I think you need OpenGL >= 2.1.

elor-arieli, 2017-01-02:
it looks like they are both version 4.4
[image]

rossant, 2017-01-02:
The "memory: 0 MB" is weird, maybe it's a config problem with the GPU. Could you check out this link: http://www.pvladov.com/2013/01/improve-intel-hd-graphics-performance.html ?

rossant, 2017-01-02:
and also http://www.pvladov.com/2013/10/increase-intel-hd-graphics-dedicated-video-memory.html

elor-arieli, 2017-01-02:
it seems that i have 128MB of dedicated memory, but i can try and add more like the guide shows.
[image]
do you still think this is the issue?

rossant, 2017-01-02:
It might be it but I'm not sure at all. 128 MB seems quite small so you can always try to increase it as much as possible and try again...

elor-arieli, 2017-01-02:
i will try.
but the main thing that bothers me is that the trace and waveforms are fine, its only the correlogram and features that are wrong.
i also checked the kwik file on another comp and all was fine, so its a display or code issue here.

rossant, 2017-01-02:
Could you send me your dataset so that I double check?
It might be that this dataset is too big for 128 MB. The feature view consumes a lot of memory.

elor-arieli, 2017-01-02:
Oh, ok. so i'll try and increase it and check again.
if it doesnt work ill send you the data.
which files do you need? all of them? (its kinda heavy..)

rossant, 2017-01-02:
Yes, all of them... You can use dropbox or similar to send the files...
Thanks

elor-arieli, 2017-01-02:
ok, so i changed video memory to 512MB, and still the same issue persists.
i think the main issue is this: OSError: exception: access violation reading 0x0000000000000008
i am attaching a link to this data set so you can look at it.
https://drive.google.com/open?id=0B2s6Lg_MNSsXakExcFZQa2lCZXc
(it says it will take another 30 min to complete the upload, its ~6000 files)

elor-arieli, 2017-01-02:
the files are ready.
i need to leave the office now (its night here), but will be back tomorrow.
thanks for the help.

elor-arieli, 2017-01-03:
any chance you looked at the files?
thanks.

rossant, 2017-01-03:
Yes, thanks for the files. It turns out there was a bug on datasets with only 1 channel. This should be fixed in [this pull request](https://github.com/kwikteam/phy/pull/732). You can try it with:

```
pip install git+https://github.com/rossant/phy@fixes --upgrade
```

elor-arieli, 2017-01-04:
Hi, 
but this is not the version i have, in the site it says to install from here:
https://github.com/kwikteam/klusta
and that this install both klustakwik2 and phy (with kwik-gui)
so i do not have the files you told me to get. (the package layout must have changed)
also when i try the command "pip install git+https://github.com/rossant/phy@fixes --upgrade"
i get this error:
Collecting git+https://github.com/rossant/phy@fixes
  Cloning https://github.com/rossant/phy (to fixes) to d:\users\ananm4\appdata\local\temp\pip-uuklqcvb-build
  Error [WinError 2] The system cannot find the file specified while executing command git clone -q https://github.com/rossant/phy D:\Users\AnanM4\AppData\Local\Temp\pip-uuklqcvb-build
Cannot find command 'git'
thanks again for the help.

rossant, 2017-01-04:
You should install [git](https://github-cloud.s3.amazonaws.com/releases/23216272/89aa6018-b7c7-11e6-8a11-43b6187be2e0.exe?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20170104%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170104T141550Z&X-Amz-Expires=300&X-Amz-Signature=996cace18b9d4b958bdfbc2ef539c39f3cfed2ecc650eb66039c9888cf5affdd&X-Amz-SignedHeaders=host&actor_id=1942359&response-content-disposition=attachment%3B%20filename%3DGit-2.11.0-64-bit.exe&response-content-type=application%2Foctet-stream) and try again (you might need to restart your computer).

elor-arieli, 2017-01-04:
What about the fact that it's not the same version? It's only klusta.
Should I install phy first?

<https://github-cloud.s3.amazonaws.com/releases/23216272/89aa6018-b7c7-11e6-8a11-43b6187be2e0.exe?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20170104%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170104T141550Z&X-Amz-Expires=300&X-Amz-Signature=996cace18b9d4b958bdfbc2ef539c39f3cfed2ecc650eb66039c9888cf5affdd&X-Amz-SignedHeaders=host&actor_id=1942359&response-content-disposition=attachment%3B%20filename%3DGit-2.11.0-64-bit.exe&response-content-type=application%2Foctet-stream>

<https://github.com/kwikteam/phy/issues/734#issuecomment-270379941>, or mute
the thread

rossant, 2017-01-04:
Don't worry about it, you already have phy installed, there's only one package named "phy". Just `activate klusta` to enable your environment and then update the phy package.

elor-arieli, 2017-01-04:
OK. Thanks.
I'll install git and try.
The wierd thing is that on my other comp everything works perfectly.

elor-arieli, 2017-01-04:
installed git, updated the phy package like you said, tried to run the
kwik-gui and got this error:

(klusta) D:\data\elec\EA-e2\EA-e2-20-12-16_161220_093756\amp-A-000>phy
kwik-gui amp-A-000.kwik
16:39:23 [E] plugin:109           No module named 'phy.cluster.manual'
Traceback (most recent call last):
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phy\utils\plugin.py",
line 107, in discover_plugins
    mod = imp.load_module(modname, file, path, descr)  # noqa
  File "D:\Users\AnanM4\Miniconda3\envs\klusta\lib\imp.py", line 234, in
load_module
    return load_source(name, filename, file)
  File "D:\Users\AnanM4\Miniconda3\envs\klusta\lib\imp.py", line 172, in
load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 693, in _load
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 665, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in
_call_with_frames_removed
  File "D:\Users\AnanM4\.phy\plugins\gui.py", line 17, in <module>
    from phy.cluster.manual.controller import Controller
ImportError: No module named 'phy.cluster.manual'
16:39:23 [E] plugin:109           No module named 'phy.cluster.manual'
Traceback (most recent call last):
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phy\utils\plugin.py",
line 107, in discover_plugins
    mod = imp.load_module(modname, file, path, descr)  # noqa
  File "D:\Users\AnanM4\Miniconda3\envs\klusta\lib\imp.py", line 234, in
load_module
    return load_source(name, filename, file)
  File "D:\Users\AnanM4\Miniconda3\envs\klusta\lib\imp.py", line 172, in
load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 693, in _load
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 665, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in
_call_with_frames_removed
  File "D:\Users\AnanM4\.phy\plugins\phycontrib_loader.py", line 3, in
<module>
    import phycontrib
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phycontrib\__init__.py",
line 15, in <module>
    from .kwik_gui import KwikGUIPlugin, KwikController  # noqa
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phycontrib\kwik_gui\__init__.py",
line 10, in <module>
    from .gui import KwikGUIPlugin, KwikController  # noqa
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phycontrib\kwik_gui\gui.py",
line 17, in <module>
    from phy.cluster.manual.controller import Controller
ImportError: No module named 'phy.cluster.manual'
Usage: phy [OPTIONS] COMMAND [ARGS]...

Error: No such command "kwik-gui".

elor-arieli, 2017-01-04:
> OK could you:
>
>    1. rename D:\Users\AnanM4\.phy to D:\Users\AnanM4\.phy~
>    2. pip install git+https://github.com/rossant/phy-contrib@fixes
>    --upgrade

cant rename
[image: Inline image 1]

elor-arieli, 2017-01-04:
i mean the problem is the ~

rossant, 2017-01-04:
Arf right windows cannot rename folders that start with a `.` :(
just rename the folder into "oldphy" or whatever, you can as well trash it but we never know

elor-arieli, 2017-01-04:
ok done, it works now but still no correlogram or feature view:
[image: Inline image 1]
and the error is still the same error from the beggining:
OSError: exception: access violation reading 0x0000000000000008
ERROR: Invoking <bound method BaseCanvas.on_draw of <CorrelogramView
(PyQt4) at 0x2c7d7f6d630>> for DrawEvent
16:50:07 [E] logs:326             Invoking <bound method BaseCanvas.on_draw
of <CorrelogramView (PyQt4) at 0x2c7d7f6d630>> for DrawEvent
WARNING: Traceback (most recent call last):
  File "D:\Users\AnanM4\Miniconda3\envs\klusta\Scripts\phy-script.py", line
11, in <module>
    load_entry_point('phy==1.0.9', 'console_scripts', 'phy')()
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\click\core.py",
line 716, in __call__
    return self.main(*args, **kwargs)
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\click\core.py",
line 696, in main
    rv = self.invoke(ctx)
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\click\core.py",
line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\click\core.py",
line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\click\core.py",
line 534, in invoke
    return callback(*args, **kwargs)
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\click\decorators.py",
line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phycontrib\kwik\gui.py",
line 459, in gui
    ctx, globals(), locals())
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phy\utils\cli.py",
line 76, in _run_cmd
    return exec_(cmd, glob, loc)
  File "<string>", line 1, in <module>
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phycontrib\kwik\gui.py",
line 435, in _run
    run_app()
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phy\gui\qt.py",
line 115, in wrapped
    return func(*args, **kwargs)
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\phy\gui\qt.py",
line 136, in run_app
    return QT_APP.exit(QT_APP.exec_())
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\vispy\app\backends\_qt.py",
line 631, in paintGL
    self._vispy_canvas.events.draw(region=None)
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\vispy\util\event.py",
line 436, in __call__
    self._invoke_callback(cb, event)
  File
"D:\Users\AnanM4\Miniconda3\envs\klusta\lib\site-packages\vispy\util\event.py",
line 452, in _invoke_callback
    self, cb_event=(cb, event))
  << caught exception here: >>
[... 147 lines cut ...]
of <FeatureView (PyQt4) at 0x2c7d7f64748>> for DrawEvent
ERROR: Invoking <bound method BaseCanvas.on_draw of <CorrelogramView
(PyQt4) at 0x2c7d7f6d630>> repeat 2
16:50:12 [E] logs:332             Invoking <bound method BaseCanvas.on_draw
of <CorrelogramView (PyQt4) at 0x2c7d7f6d630>> repeat 2
ERROR: Invoking <bound method BaseCanvas.on_draw of <FeatureView (PyQt4) at
0x2c7d7f64748>> repeat 2
16:50:12 [E] logs:332             Invoking <bound method BaseCanvas.on_draw
of <FeatureView (PyQt4) at 0x2c7d7f64748>> repeat 2
ERROR: Invoking <bound method BaseCanvas.on_draw of <CorrelogramView
(PyQt4) at 0x2c7d7f6d630>> repeat 3
16:50:17 [E] logs:332             Invoking <bound method BaseCanvas.on_draw
of <CorrelogramView (PyQt4) at 0x2c7d7f6d630>> repeat 3
ERROR: Invoking <bound method BaseCanvas.on_draw of <FeatureView (PyQt4) at
0x2c7d7f64748>> repeat 3
16:50:17 [E] logs:332             Invoking <bound method BaseCanvas.on_draw
of <FeatureView (PyQt4) at 0x2c7d7f64748>> repeat 3
ERROR: Invoking <bound method BaseCanvas.on_draw of <FeatureView (PyQt4) at
0x2c7d7f64748>> repeat 4
16:50:49 [E] logs:332             Invoking <bound method BaseCanvas.on_draw
of <FeatureView (PyQt4) at 0x2c7d7f64748>> repeat 4
ERROR: Invoking <bound method BaseCanvas.on_draw of <CorrelogramView
(PyQt4) at 0x2c7d7f6d630>> repeat 4
16:50:49 [E] logs:332             Invoking <bound method BaseCanvas.on_draw
of <CorrelogramView (PyQt4) at 0x2c7d7f6d630>> repeat 4

rossant, 2017-01-04:
ok give me your email it'll be simpler

elor-arieli, 2017-01-04:
elor.arieli@gmail.com
thanks again for the help
this is very helpful and we need it for our lab.

rossant, 2017-01-23:
This was apparently due to a bug in the Intel graphics driver. The problem was solved by using an nvidia GPU instead.

closed 2017-01-23 by rossant