cortex-lab/phy #737: bug when launching template-gui through Windows remote desktop

dshimaoka, 2017-01-13:
When template-gui is launched through Windows remote desktop, there appears many error messages in command line, and traces do not appear in any of the views. This does not happen when template-gui is launched locally and then accessed through remote desktop.

rossant, 2017-01-15:
I think this is related: https://www.opengl.org/discussion_boards/showthread.php/198132-OpenGL-and-Microsoft-Remote-Desktop

marius10p, 2019-02-03:
I looked more [into this](https://social.technet.microsoft.com/Forums/windowsserver/en-US/c8295ef8-3711-4576-9293-2c4965280165/opengl-and-remote-desktop?forum=winserverTS) and it seems that a workaround is to start the opengl program before the remote session. I had Phy already running and remote-ing allowed the graphics session to continue fine. I then tried connecting with teamviewer again and this time Phy started with full graphics capabilities. 

So my workaround is to start Phy from teamviewer, and then switch to remote desktop. Or just use teamviewer.

closed 2019-06-13 by rossant