cortex-lab/phy #741: bug: similarity values are lost when splitting

nsteinme, 2017-01-31:
This is very important - the two parts of the split should maintain the similarity values they had with all existing clusters!! So if the pre-split cluster was showing up as a near match, the split version will too. Otherwise they're just lost...

nsteinme, 2017-01-31:
I think the values are kept properly - it is just that for some reason the other cluster (one you didn't split) doesn't see either of the split clusters in its SimilarityView. 

To try to be more clear:

I select A and see that B is at the top of A's SimiliarityView, and vice versa. I split B, creating C and D. A is at the top of C's and D's SimiliarityView, but neither C nor D are present anymore in A's. They need to be there in A's SimiliarityView.

nsteinme, 2017-01-31:
Ok, further observation, this is not unique to splits - also happens with merges. It is some general failure to update similarityview when the clustering changes. I think this view just needs to be refreshed somehow, but it isn't refreshed even if restart the program. It does get "refreshed" if I close the program and delete the .phy directory though. So this seems to confirm again that the similarity scores are updates properly, but the similarityview comes from a cached version that needs to be updated when merges/splits happen.

rossant, 2017-02-01:
Thanks for doing half of the work for me. This should be fixed on phycontrib's master.

rossant, 2017-02-02:
I think this is fixed, feel free to reopen if it is not.

closed 2017-02-02 by rossant