cortex-lab/phy #744: singular matrix error when importing from kilosort

jerryhanson, 2017-03-08:
Hi there,

An error emerged when I import kilosort data to template-gui. Could anyone help me with this?

I could import eMouse simulated data(by using kilosort) successfully.

I am also pasting files' link of whitening_mat.npy and whitening_mat_inv.npy.
[https://1drv.ms/f/s!AkTnDbL6As39gYh51YSaEFsL-vCq3g](url)
 
Following scrips are pulled from phy.log

2017-03-08 15:36:18 [D] gui:268              Loading amplitudes.
2017-03-08 15:36:18 [D] gui:277              Loading 30339 spike clusters.
2017-03-08 15:36:18 [D] gui:283              Loading spike templates.
2017-03-08 15:36:18 [D] gui:289              Loading spike samples.
2017-03-08 15:36:18 [D] gui:293              Loading templates.
2017-03-08 15:36:18 [D] gui:299              Loading the whitening matrix.
2017-03-08 15:36:18 [D] gui:307              Couldn't find unwhitened templates, computing them.
2017-03-08 15:36:18 [D] gui:309              Inversing the whitening matrix (256, 256).
**2017-03-08 15:36:18 [E] cli:39               An error has occurred (LinAlgError): Singular matrix**
2017-03-08 15:36:18 [D] cli:42               Traceback (most recent call last):
  File "c:\users\user\miniconda3\envs\phy\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\user\miniconda3\envs\phy\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\user\Miniconda3\envs\phy\Scripts\phy.exe\__main__.py", line 9, in <module>
    sys.exit(phy())
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\click\core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\click\core.py", line 697, in main
    rv = self.invoke(ctx)
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\click\core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\click\core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\click\core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\phycontrib\template\gui.py", line 814, in cluster_manual
    _run_cmd('_run(params)', ctx, globals(), locals())
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\phy\utils\cli.py", line 76, in _run_cmd
    return exec_(cmd, glob, loc)
  File "<string>", line 1, in <module>
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\phycontrib\template\gui.py", line 786, in _run
    controller = TemplateController(**params)
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\phycontrib\template\gui.py", line 249, in __init__
    super(TemplateController, self).__init__()
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\phy\cluster\manual\controller.py", line 67, in __init__
    self._init_data()
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\phycontrib\template\gui.py", line 310, in _init_data
    wmi = np.linalg.inv(self.whitening_matrix)
  File "c:\users\user\miniconda3\envs\phy\lib\site-packages\numpy\linalg\linalg.py", line 526, in inv
    ainv = _umath_linalg.inv(a, signature=signature, extobj=extobj)
  **File "c:\users\user\miniconda3\envs\phy\lib\site-packages\numpy\linalg\linalg.py", line 90, in _raise_linalgerror_singular
    raise LinAlgError("Singular matrix")
numpy.linalg.linalg.LinAlgError: Singular matrix`**

rossant, 2017-03-08:
Hi, could you send me the full dataset, or the dataset corresponding to a small excerpt of the data that reproduces the problem? Thanks

jerryhanson, 2017-03-08:
I am uploading my *.dat and generated *.npy files into onedrive disk. Let me know if you cannot download from there. This is a one minute recording sample.

https://1drv.ms/f/s!AkTnDbL6As39gYh9vbpUs5Qd2kPkLQ

I don't know if this is relevant. When I was running this line `rez.WrotInv         =(rez.Wrot/200)^-1;`during kilosort process, matlab always warned me of **infinite matrix**. And the rez.WrotInv was actually becoming a infinite matrix. After I changed this line to `rez.WrotInv         =1./ (rez.Wrot/200);`, this warning went away.

I think very likely this **singular matrix error** was caused by **parameter setting up for whitening process**. So I am also uploading the **standardConfig.m** file used in kilosort. Please pay special attention to the parameters for whitening process. Let me know if you find something suspicious.

rossant, 2017-03-08:
OK I'll have a look. I put @marius10p in CC in case he has suggestions regarding kilosort.

marius10p, 2017-03-27:
Sorry, I just noticed this. You need to take the full inverse of the matrix: your version replaces this with an element-wise inverse, which is a completely different matrix. This error could happen if one of your channels is completely dead, or if two channels are exactly the same. 

So it might indicate something else that is wrong with the data or the configuration. If you want to ignore this error anyway, add a very small diagonal matrix before doing the inversion.

jerryhanson, 2017-03-28:
hi, @marius10p . I close this issue because I already solved it by splitting 256 channel file into eight files with 32 channels in each one. In this way, I just leaved the standardConfig file the same parameter as I downloaded from github archive. I could import data to phy successfully without warning alert. Another reason for splitting file is because the RAM size of our computer won't allow us to sort 256 channel data file in a single run. Thanks for your reply anyway.

After inporting kilosorted data to phy, there seems something wired with the displaying in the trace view. I mean these gray vertical lines. Is this normal, or what might be causing it? @rossant 

screen snapshot
https://1drv.ms/i/s!AkTnDbL6As39gYh-EdLDAfwWRhl29w

rossant, 2017-03-29:
That's weird, what happens if you change the y-scaling in the trace view using alt+up/down?
Do you have the latest version of the graphics driver installed?

closed 2019-06-13 by rossant