cortex-lab/phy #746: WaveformView conundrum

brunoEm, 2017-03-21:
Hi,

I'm running phy-1.0.9. Is there any reason why one would see not one, but 16 channels highlighted by the cluster color in WaveformView on all the putative units? Is this some feature of this version or does that indicate something is amiss?

rossant, 2017-03-22:
Are you using the kwik GUI or the template GUI?

brunoEm, 2017-03-22:
Template GUI.

rossant, 2017-03-22:
At the moment a fixed number of channels is chosen per cluster: the channel with the highest amplitude waveform, and its N neighbors. We'll improve this when we'll implement sparse export in kilosort. If you use spyking circus instead, the sparse export is already implemented and it's spyking circus that chooses the best channels per cluster.

brunoEm, 2017-03-22:
Thanks a lot!

nikolaskaralis, 2017-03-22:
On a related note:
I am trying phy with the artificial data from KiloSort and there is something confusing about the channels numbers.

When clicking on a cluster, if I understand correctly the channel number shown is the channel with max amplitude. However, in the waveform view, although indeed the highlighted channels are centered around the channel with max amplitude, the actual numbers shown are very different and do not correspond to the number shown in the cluster view.

Is there something I am doing wrong?

Below a screenshot:
[image]

marius10p, 2017-03-22:
Hi @rossant , what do you mean by "sparse export" ? Do you need Kilosort to threshold which channels should be highlighted? I thought this would be done in Phy. 

@nikolaskaralis Looks like something to do with the channel map. @rossant, are the numbers shown in the waveform view the indices in the binary file, or the indices from the channel map?

rossant, 2017-03-22:
Yes at some point it would be good if the export script in kilosort could output sparse arrays for templates, like in spyking circus. It's just a post-processing step to generate the npy files used by phy.

The channel map bug might be fixed on the development version of phy and phycontrib, feel free to try it and let me know if it still doesn't work.

marius10p, 2017-03-22:
@rossant How should the sparse arrays be formatted? Can you please open an issue on Kilosort and give the details of the required formatting there? Thanks.

nikolaskaralis, 2017-03-22:
@rossant  Indeed it works on the development version. Thanks!

phrenico, 2017-07-20:
@nikolaskaralis @rossant  And which number was actually the correct channel index? (on the channel map or on the cluster view)

closed 2019-06-13 by rossant