cortex-lab/phy #747: Problem with paths

nikolaskaralis, 2017-03-21:
I am trying to use phy on mac and I experience the following problem:
When providing the absolute or relative path to the params file, it doesn't work and I get the following
23:35:12 [W] gui:264              Error while loading data: File sim_binary.dat not found.
23:35:12 [E] cli:39               An error has occurred (FileNotFoundError): [Errno 2] No such file or directory: 'amplitudes.npy'

However, when I change into the directory where params.py is contained, and I run: 
phy template-gui params.py 

then it works.

Is this the intended behavior?

rossant, 2017-03-22:
Have you tried to put the absolute path to the dat file in the params file?

nikolaskaralis, 2017-03-22:
Hi Cyrille,
this also doesn't work.

The error about the binary file missing stops occurring, but still, the
error about the amplitude file occurs:
No such file or directory: 'amplitudes.npy'

Still, everything works when changing to the directory.

rossant, 2017-03-22:
Can you try adding this in the Python params file?

`dat_path='/path/to/dir_with_npy_files/'`

nikolaskaralis, 2017-03-22:
It complains it is a directory.
I guess phy searches for the .npy file in the local folder and not the
subfolder where sim_binary.dat is contained.

12:54:00 [E] cli:39               An error has occurred
(IsADirectoryError): [Errno 21] Is a directory:
'/storage/nikolas/data/temp/testKS/'

My suggestion would be that in the terminal one can provide a path to the
params.py file and then phy looks for a .dat file and amplitudes.npy file
inside this folder and uses them.

rossant, 2017-03-22:
Sorry I meant `dir_path` not `dat_path`...

yger, 2017-03-22:
It works with dir_path. The script used by SpyKING CIRCUS to launch the phy
gui uses that, and it can be launched from outside the folder where the
.npy are located

Pierre

closed 2019-06-13 by rossant