cortex-lab/phy #748: Cluster splitting from TemplateFeatureView

nikolaskaralis, 2017-03-22:
In many instances, it is much easier to split a cluster on the TemplateFeatureView than on the FeatureView. It would be great if the polygon could be drawn on the TemplateFeatureView as well (now it is only possible on the FeatureView).

nikolaskaralis, 2017-03-27:
On a related note:
Is it possible to have the possibility of drawing connected lines instead of polygons?
The end result would be the same, but the issue with the current method is that the polygon is closed from the beginning which makes it very counterintuitive to create fine polygons.
What I am proposing is that for each click a line connecting the last point with the new one is formed, but the polygon remains open until you click again on the first node.

ereming, 2017-04-20:
I also have found instances where splitting appears to be straightforward in the TemplateFeatureView. Here is a screenshot of one:
[image]
There are many spikes assigned to the red cluster which appear to fit better with the blue cluster (or maybe with both) in the TemplateFeatureView. I don't see a similar boundary in the FeatureView.

greggheller, 2017-09-15:
I've noticed this as well and would be a third person to support splitting in the TemplateFeatureView.

rossant, 2019-06-13:
done in dev branch

closed 2019-06-13 by rossant