cortex-lab/phy #753: Segmentation fault when saving

malfatti, 2017-06-01:
Hello!

After `klusta MyRec.prm`, I can load the GUI with `phy kwik-gui MyRec.kwik` and play w/ the data, however, when I press `Ctrl+S` I get a segmentation fault, with no other apparent error in debug mode, or in the log.

I can reproduce this with Klusta example data:
```bash
$ pip install git+https://github.com/kwikteam/phy git+https://github.com/kwikteam/phy-contrib --upgrade
[...]
Successfully installed phy-1.0.9 phycontrib-1.0.15

$ klusta hybrid_10sec.prm
[...]
22:17:39 [I] klustakwik Trying to split clusters
22:17:40 [I] klustakwik No points changed, previous step was full and did not split, so finishing.
22:17:40 [I] launch:131           The automatic clustering process has finished.
22:17:40 [I] launch:268           Save the clustering to `hybrid_10sec.kwik`.
22:17:40 [I] model:1085           Save the Kwik file at `/home/malfatti/TestKlusta/hybrid_10sec.kwik`.
22:17:40 [I] launch:290           Clustering done!

$ phy kwik-gui hybrid_10sec.kwik --debug
# GUI loads correctly, besides the Qt warning
QGtkStyle could not resolve GTK. Make sure you have installed the proper libraries.
22:25:11 [D] model:671            Enable waveform filter.
22:25:11 [D] model:467            Loading traces: hybrid_10sec.dat
22:25:11 [D] model:1067           Kwik file `/home/malfatti/TestKlusta/hybrid_10sec.kwik` loaded.
22:25:11 [D] context:67           Initialize joblib cache dir at `/home/malfatti/TestKlusta/.phy`.
22:25:11 [D] context:91           Load memcache for `phycontrib.kwik.gui.get_best_channels`.
22:25:11 [D] context:91           Load memcache for `phycontrib.kwik.gui.get_probe_depth`.
22:25:11 [D] context:91           Load memcache for `phycontrib.kwik.gui._get_mean_masks`.
22:25:11 [D] context:91           Load memcache for `phycontrib.kwik.gui._get_mean_waveforms`.
22:25:11 [D] context:148          The file `/home/malfatti/TestKlusta/.phy/new_cluster_id.pkl` doesnt exist.
22:25:11 [D] context:134          Save data to `/home/malfatti/TestKlusta/.phy/spikes_per_cluster.pkl`.
22:25:11 [D] config:46            Load config file `/home/malfatti/.phy/phy_config.py`.
22:25:11 [D] gui:432              Load the GUI state from `/home/malfatti/.phy/KwikGUI/state.json`.
22:25:11 [D] gui:173              Load the geometry state.
# Then I just selected a cluster and pressed Ctrl-S
22:25:13 [D] supervisor:342       Select cluster(s): 0.
Segmentation fault

$ cat phy.log
2017-05-31 22:25:11 [D] model:671            Enable waveform filter.
2017-05-31 22:25:11 [D] model:467            Loading traces: hybrid_10sec.dat
2017-05-31 22:25:11 [D] model:1067           Kwik file `/home/malfatti/TestKlusta/hybrid_10sec.kwik` loaded.
2017-05-31 22:25:11 [D] context:67           Initialize joblib cache dir at `/home/malfatti/TestKlusta/.phy`.
2017-05-31 22:25:11 [D] context:91           Load memcache for `phycontrib.kwik.gui.get_best_channels`.
2017-05-31 22:25:11 [D] context:91           Load memcache for `phycontrib.kwik.gui.get_probe_depth`.
2017-05-31 22:25:11 [D] context:91           Load memcache for `phycontrib.kwik.gui._get_mean_masks`.
2017-05-31 22:25:11 [D] context:91           Load memcache for `phycontrib.kwik.gui._get_mean_waveforms`.
2017-05-31 22:25:11 [D] context:148          The file `/home/malfatti/TestKlusta/.phy/new_cluster_id.pkl` doesnt exist.
2017-05-31 22:25:11 [D] context:134          Save data to `/home/malfatti/TestKlusta/.phy/spikes_per_cluster.pkl`.
2017-05-31 22:25:11 [D] config:46            Load config file `/home/malfatti/.phy/phy_config.py`.
2017-05-31 22:25:11 [D] gui:432              Load the GUI state from `/home/malfatti/.phy/KwikGUI/state.json`.
2017-05-31 22:25:11 [D] gui:173              Load the geometry state.
2017-05-31 22:25:13 [D] supervisor:342       Select cluster(s): 0.
```

The kwik file loads ok in ipython, but no changes were saved to the file.
Does anyone has/had this problem?

Cheers

rossant, 2017-06-01:
Hi, that's weird. Does it crash if you don't select any cluster before saving? What is the version of all of your Python packages in your environment (in particular h5py)?

malfatti, 2017-06-01:
Hej :) Yes, it crashes just opening and saving, with almost the same log (except the "Select cluster(s)" line).

I installed Klusta through conda, according to instructions in the README.md, and upgraded phy to development version. Here are the *-info files:
```bash
$ cd $KlustaEnv/lib/python3.5/site-packages/ 

$ ls *info -d
click-6.7.dist-info
cookies-2.2.1.dist-info
cycler-0.10.0-py3.5.egg-info
Cython-0.25.2-py3.5.egg-info
decorator-4.0.11-py3.5.egg-info
h5py-2.7.0-py3.5-linux-x86_64.egg-info
ipython-6.0.0-py3.5.egg-info
ipython_genutils-0.2.0-py3.5.egg-info
jedi-0.10.2-py3.5.egg-info
joblib-0.11.dist-info
klusta-3.0.16.dist-info
klustakwik2-0.2.6.dist-info
matplotlib-1.5.1-py3.5.egg-info
numpy-1.11.3-py3.5.egg-info
pexpect-4.2.1-py3.5.egg-info
phy-1.0.9-py3.5.egg-info
phycontrib-1.0.15-py3.5.egg-info
pickleshare-0.7.4-py3.5.egg-info
pip-9.0.1-py3.5.egg-info
prompt_toolkit-1.0.14-py3.5.egg-info
ptyprocess-0.5.1-py3.5.egg-info
py-1.4.33.dist-info
Pygments-2.2.0-py3.5.egg-info
PyOpenGL-3.1.0.dist-info
pyparsing-2.1.4-py3.5.egg-info
pytest-3.0.7.dist-info
python_dateutil-2.6.0-py3.5.egg-info
pytz-2017.2-py3.5.egg-info
requests-2.14.2.dist-info
responses-0.5.1.dist-info
scipy-0.19.0-py3.5.egg-info
simplegeneric-0.8.1-py3.5.egg-info
six-1.10.0-py3.5.egg-info
tqdm-4.11.2.dist-info
traitlets-4.3.2-py3.5.egg-info
vispy-0.4.0.dist-info
wcwidth-0.1.7-py3.5.egg-info
wheel-0.29.0-py3.5.egg-inf
```

Thanks for your response :D

malfatti, 2017-06-02:
Hej again,

I tried the simplest of things: moved my `~/.gtkrc-2.0` and `~/.config/{Trolltech.conf,gtk-2.0,gtk-3.0}` files to a backup folder and... voilà! The QGtkStyle warning disappeared and data is saved without errors.

I would like to ask you to keep this issue open for now. In the next days I'll investigate what setting exactly will generate this segmentation fault. Then, I'll report back and we close it.

Thanks and sorry for the noise :)

rossant, 2017-06-02:
Happy to hear that!

malfatti, 2017-06-28:
Hej! Sorry for the huge delay.

Just to report, in case someone have the same problem. After some digging it turns out that, at least in my gentoo system, using any GTK theme as QT theme (using qt5ct) makes the segmentation fault to happen again. Choosing "Fusion", for example, fix it.

Cheers :)

closed 2017-06-28 by malfatti