cortex-lab/phy #756: Mapping from cluster_id to channel and/or depth?

PhantomSpike, 2017-06-28:
Hi,

This should be a quick and straightforward to answer:
Is there a way to find which cluster_id corresponds to what depth and have this information in a .mat or .npy file?

In the phy gui this information is readily available in the ClusterView but I would like to have it tabulated in some more useful format such as .mat, .npy, .csv etc.. This is because I want to be able to know which spike times belonging to a particular cluster came from what depth/channel and for example plot what is the distribution of responses along the recording depth.

Any suggestions would be greatly appreciated!

Best wishes,

Alex

rossant, 2017-06-28:
Hi
you can obtain this information in Python and export to a CSV. Here is a code snippet that should work (the API may change in the future):

```python
from phycontrib.template import TemplateController
from phy.utils._misc import _read_python
tc = TemplateController(**_read_python('/path/to/params.py'))
print(tc.get_best_channel(10))  # print the best channel for cluster #10
tc.model.get_template(10)  # gives an object with the best channel, the template waveform...
```

There is no doc at the moment but you can [look at the code](https://github.com/kwikteam/phy-contrib/blob/master/phycontrib/template/) or use IPython to introspect the model and controller and see the available methods.

PhantomSpike, 2017-12-04:
Hi rossant,

I have finally come around to try this.

When I run your code I get an error on the line:
tc = TemplateController(**_read_python('/path/to/params.py'))

The error is the following:
14:15:35 [E] cli:39               An error has occurred (EOFError): Ran out of input

What would you recommend? I have the latest version of phy which works fine when using for manual spike sorting and the associated GUI. I also have the latest version of miniconda3.

Many thanks for your help!

Best wishes,

Alex

rossant, 2017-12-06:
Have you replaced `'/path/to/params.py'` by the path to your own params.py file?

PhantomSpike, 2017-12-13:
Hey,

Thanks for your reply.

Yes, I have, so I think the problem lies elsewhere.

I am relatively new to Python though, so it could be also programming incompetence on my side. Is it possible to convert the file that contains the depth info into a Matlab one and manipulated within Matlab? I am more familiar with it.

rossant, 2019-06-13:
will be closed by an upcoming feature where all cluster view information is going to be automatically saved in a tsv file

closed 2019-06-13 by rossant