cortex-lab/phy #759: 3D probe summary view

nsteinme, 2017-07-20:
@rossant, is it possible to make scatter plots with 3 dimensions in phy? If so, I have a suggestion for one that could make a really nice probe overview. Here's how it works: 

For each spike, compute two things: 1) its peak channel (i.e. channel with largest waveform), and take its PC1 and PC2 from that channel; 2) its position on the probe in the y-dimension, computed as a center of mass of its PC1 projections across channels. Then the plot is: PC1_peakChan vs PC2_peakChan vs yPosition. Coloration could be with a unique color per cluster, or only coloring the selected ones - just as in traceview, basically. 

@marius10p this is a sort of "shortcut" version of the t-sne idea we talked about earlier today. Do you think it makes sense?

@rossant let me know if 3D scatters are possible. 

If both of you come back positively, I will make a version in matlab to check that it actually looks reasonable and useful (maybe not, but let's see), then I'll come back here with a more detailed description of what exactly to plot if it looks good.

rossant, 2017-07-21:
3D plots are not implemented at the moment but I can do it if that's really necessary. The only caveat is that 3D typically requires user interaction which takes more time than just looking at static figures. Or we can animate it by default. Just something to keep in mind.

nsteinme, 2017-07-24:
Ok, I tried this briefly, but sticking to a 2D view for the reasons @rossant has given. Actually it was very simple to make and produces an attractive and useful plot: 

[image]

I think this is useful because we currently don't have any other way to just "see everything around here". 

To get this plot I first estimate the position of each spike, which I do by taking a center of mass of the PC's (first PC only), exact code below. The x-axis is just the value of the first PC on whatever channel is declared as the top channel for that spike. 

I guess the only real question is how to handle the large number of points that would be present, like whether there's a way to adaptively add more points as the user zooms in. Or if not, maybe representing the same space as a density plot would be better. 

```
pcFeat = squeeze(pcFeat(:,1,:)); % take first PC only
pcFeat(pcFeat<0) = 0; % some entries are negative, but we don't really want to push the CoM away from there. 

% which channels for each spike? 
spikeFeatInd = pcFeatInd(spikeTemps+1,:); % +1 is because matlab uses one-indexing

% ycoords of those channels?
spikeFeatYcoords = ycoords(spikeFeatInd+1);

% center of mass is sum(coords.*features)/sum(features)
spikeDepths = sum(spikeFeatYcoords.*pcFeat.^2,2)./sum(pcFeat.^2,2); % using features squared, I just thought it worked better

xData = pcFeat(:,1); % that's it - just take the first channel
```

rossant, 2019-06-13:
can be done as a spike attribute in the dev branch

closed 2019-06-13 by rossant