cortex-lab/phy #760: Raw voltage value spike Amplitude - Amplitude/Waveform View

Sarruedi, 2017-07-20:
Is is possible to visualize the actual voltage for the spike amplitude view as well as the waveform view ion phy? The default view does not contain any information on the actual raw amplitudes. In addition, the amplitude view seems to not rescale when one merges clusters. I would like to see the raw amplitude or some sort of voltage scale bar to use as an additional measure for isolating good units and to see to the difference in spike amplitudes of bursting neurons.

greggheller, 2017-09-15:
I would like to second this. 

My understanding was that the amplitude view shows the amount the template was scaled by to fit the raw data. Thus if the templates were different sizes the same spike would require different amounts of scaling, and after merging the amplitude view would appear bimodal. Is this interpretation correct? Is there any way to get the raw amplitudes, even in histogram form?

I'm trying out nicks plugin (https://github.com/LBHB/phy-contrib/blob/master/phycontrib/LBHB_plugins/AmplitudeHistogram.py) but I think it uses the same scaling factors as amplitude values: coords = controller._get_amplitudes(clusters[i])

I'm not much of a coder and new to python, so I'm still trying to figure things out.

rossant, 2019-06-20:
closed by #842

closed 2019-06-20 by rossant