cortex-lab/phy #762: Channel order scrambled when sorting

karenw13, 2017-07-26:
Any time separate clusters in a channel are grouped or marked as multi/good unit or noise, the program scatters the order of the channels so they are arranged by id # instead of channel #. Is there a way to fix this?Also, occasionally when sorting Kilosort stops automatically changing the channel you are viewing in WaveformView so that it just stays on the same channel, showing noise, as you click through others in ClusterView.

nikolaskaralis, 2017-08-20:
Yes indeed the order updating is very distracting and slows down progress

greggheller, 2017-09-15:
I have had a similar problem as well. I like to look through clusters pre-categorized as noise and every time I move one out of the noise category I have to go back and find where I was.

Is there any way to change this?

rossant, 2019-06-13:
should be fixed in the dev branch

closed 2019-06-13 by rossant