cortex-lab/phy #767: FeatureView

mjk1028, 2017-08-12:
Dear,

i have been using phy but from yesterday i do not load (see) data in featureview. i might press something wrong but not sure what causes this and how to go back to the usual setup. i reinstall phy but did not help. do you recognize this and how to load data properly?

thank you so much for your help in advance!

best regards,
MJ

[image]

rossant, 2017-09-08:
can you check that the pc_features.npy files are present in the directory?

irisbhy, 2018-02-15:
Hello,
I am having a similar issues with some of my features not displaying properly, including FeatureView and CorrelogramView0. If you were able to solve this problem, please let me know!
and pc_features.npy is in fact in my directory.

Thank you,
Iris

rossant, 2019-06-13:
please reopen if the problem persists in the dev branch

closed 2019-06-13 by rossant