cortex-lab/phy #768: JOBLIB issue

whitepine, 2017-08-24:
I get the following error when trying to run the phy kwik-gui command. I only get this issue when I try to run this command on a mapped (or remote) network drive. My machine is running Windows 10.

//// error ////
anaconda3\envs\klusta\lib\site-packages\phy\cluster\manual\controller.py:413: UserWarning: Persisting input arguments took 1.17s to run.
If this happens often in your code, it can cause performance problems
(results will be correct in all cases).
The reason for this is probably some large input arguments for a wrapped
 function (e.g. large strings).
THIS IS A JOBLIB ISSUE. If you can, kindly provide the joblib's team with an
 example so that they can fix the problem.

rossant, 2017-09-07:
It seems like a warning because of latency with remote data access. Maybe you could try to delete the .phy subfolder (remote), and create there a symlink named .phy to a local folder. That might improve speed.

whitepine, 2017-09-07:
thanks I check that out!

closed 2019-06-13 by rossant