cortex-lab/phy #769: RuntimeError: context could not be created

whitepine, 2017-08-24:
I running into the following issue when running: 'phy kwik-gui my_file.kwik'

RuntimeError: context could not be created

It looks like the error when trying to load the GUI. This is the output from the console when running the debug option right before the error is thrown.

22:45:30 [D] gui:411              Load the GUI state from `/home/jthermiz/.phy/KwikGUI/state.json`.
22:45:50 [E] cli:37               An error has occurred (RuntimeError): context could not be created
22:45:50 [D] cli:40               Traceback (most recent call last):

rossant, 2017-09-07:
Seems like a problem with the GPU driver, or maybe you're running this remotely (which doesn't work)?

whitepine, 2017-09-07:
I controlling a machine remotely though remote desktop, but the command is being run locally on the machine.

Any additional insight in how it may be a GPU driver issue? I have 'mesa-utils' installed and ran 'glxgears' and 'glxinfo', which both run fine. For reference, I also posted about the google groups:

https://groups.google.com/forum/#!searchin/klustaviewas/RuntimeError$3A$20context$20could$20not$20be$20created$20%7Csort:relevance/klustaviewas/F_2p4XDx1Fc/cqcU_Z8zBwAJ

rossant, 2017-09-07:
Maybe a conflict with the remote desktop software? Do you have a way to test directly on the machine? Using OpenGL via remote desktop causes problems sometimes.

whitepine, 2017-09-07:
I can give it a try. thanks for the help!

wehrlab, 2018-08-09:
Hi, i am having similar issue with using the gui remotely. Has there been a fix for this or is it still not compatible with remoting in. Thank you.

wehrlab, 2018-08-10:
Just wanted to add a few things I have learned. The issue is caused by opening up a program that uses Nvidia graphics card from a remote computer that doesnt have it  (I think). I found a work around by writing a batch file that has a timeout of a few seconds during which I disconnect my remote desktop, let phy start while remote desktop is disconnected and reconnecting after. This is only an issue when you try to open phy remotely, once it's running it works well via remote desktop. Other remote control software can work around this issue, but some of them cost money. TigerVPN is a free software that should not have that issue but I havent tested it.

closed 2019-06-13 by rossant