cortex-lab/phy #770: Some templates rescale after a merge

greggheller, 2017-09-15:
Hi guys,

I've been using phy for a few months and it's been awesome. Nice and fast and functional. Thank you for all your work on this.

Just a small thing that I am curious about. Most of the time It is clear that phy does not rescale waveforms or templates after a merge, but I have noticed that sometimes it does, or at least appears to. In the screenshot before the merge the templates are clearly different amplitudes on the peak channel 117. After the merge they are identical. Whats going on? 

The amplitude view doesn't show any re-scaling. The first screenshot is definitely templates and not mean waveforms (thanks for adding the mean waveforms by the way).  I have observed this multiple times from multiple different recordings. So what am I actually looking when I view the template in the WaveformView window? 

I have been using the developers version but haven't updated it in a few weeks. 

[image]
[image]

zeeMonkeez, 2018-01-29:
I struggled with this, too. Not sure whether my explanation is correct, but from looking at the code I think this is what's going on:

When clusters are merged or split, templates do not get recomputed. To find the template to be shown in the plot, first all spikes in a cluster are examined for their *original* template, and numbers of spikes in each template are counted. Then the template with the most spikes 'wins', and this is what's used for template waveform display, 'best channel', 'depth', etc.

greggheller, 2018-03-19:
Hey,

Thanks for the reply!

I believe you are right as far as the template with the most spikes winning for best channel and depth. However in waveform display BOTH templates are displayed. I am (Now) pretty sure that the templates are displayed scaled by the average amplitude, so if you merge a low amplitude cluster with a high amplitude one, the average amplitude will change and the displayed template will be scaled differently after the merge.

closed 2018-03-19 by greggheller