cortex-lab/phy #772: Feature request: keep ordering in ClusterView after multiple merges or good/mua/noise classification

weallen, 2017-09-21:
Right now, in the version of phy and phy-contrib I'm using (1.0.9 and 1.0.15), regardless of what attribute I sort the clusters by (e.g. channel, depth, n_spikes) whenever a cluster is marked as good/mua/noise all the clusters are resorted by ID. So it's not currently possible to move through the clusters by n_spikes or depth and mark their type. 

Also, if more than 2 clusters are merged together the program loses your current position ClusterView, so it is not possible to progress through all the units in any order while performing multiple merges. This makes it difficult to progressively go through the whole list and do all the merges in one pass.

weallen, 2017-09-21:
(Just noticed this is a repeated of Issue #762)

rossant, 2019-06-13:
should be fixed in the dev branch

closed 2019-06-13 by rossant