cortex-lab/phy #775: Error: No such command "template-gui"

ingridvanwelie, 2017-10-20:
Hi,

I am new to Python, but need to use Kilosort with Phy. I am trying to install it on a Mac running OS Sierra and installed Miniconda 3 and phycontribute and am able to activate phy. 

When I try and run the template GUI however using "phy template-gui params.py" I get the following error message:

Error: No such command "template-gui".

When I do "phy --debug" I get the following feedback:

phy) Ingrids-MacBook-Pro:FAKEDATA lawelie$ phy --debug
13:13:32 [D] config:46            Load config file `/Users/lawelie/.phy/phy_config.py`.
Usage: phy [OPTIONS] COMMAND [ARGS]...

  By default, the `phy` command does nothing. Add subcommands with plugins
  using `attach_to_cli()` and the `click` library.

Options:
  --version   Show the version and exit.
  -h, --help  Show this message and exit.

Do you have any suggestions on why the template-GUI command is not working?

best wishes,
Ingrid

rossant, 2017-10-23:
Hi, could you activate the phy environment and do `python -c "import phy; phycontrib"` and see if there is any error message?

ingridvanwelie, 2017-10-23:
Here’s what it says: 

(phy) Ingrids-MBP:~ lawelie$ import phy; phycontribute
-bash: import: command not found
-bash: phycontribute: command not found

rossant, 2017-10-23:
Sorry, I got it wrong. In the terminal, write:

```bash
python -c "import phy,phycontrib"
```

ingridvanwelie, 2017-10-23:
> Sorry, I got it wrong. In the terminal, write:
>
> python -c "import phy,phycontrib"

Still says:

(phy) Ingrids-MBP:~ lawelie$ import phy, phycontrib
-bash: import: command not found

rossant, 2017-10-23:
you need to put the `python -c "` before

ingridvanwelie, 2017-10-23:
Like this?

(phy) Ingrids-MBP:~ lawelie$ -c import phy, phycontrib
-bash: -c: command not found

rossant, 2017-10-23:
No it should really  be:

```bash
(phy) Ingrids-MBP:~ lawelie$ python -c "import phy, phycontrib"
```

ingridvanwelie, 2017-10-23:
Ah, many thanks, that clarifies things. Here’s what I get:

(phy) Ingrids-MBP:~ lawelie$ python -c "import phy, phycontrib"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/lawelie/miniconda3/envs/phy/lib/python3.5/site-packages/phycontrib/__init__.py", line 15, in <module>
    from .kwik_gui import KwikGUIPlugin, KwikController  # noqa
  File "/Users/lawelie/miniconda3/envs/phy/lib/python3.5/site-packages/phycontrib/kwik_gui/__init__.py", line 10, in <module>
    from .gui import KwikGUIPlugin, KwikController  # noqa
  File "/Users/lawelie/miniconda3/envs/phy/lib/python3.5/site-packages/phycontrib/kwik_gui/gui.py", line 17, in <module>
    from phy.cluster.manual.controller import Controller
  File "/Users/lawelie/miniconda3/envs/phy/lib/python3.5/site-packages/phy/cluster/__init__.py", line 6, in <module>
    from . import manual
  File "/Users/lawelie/miniconda3/envs/phy/lib/python3.5/site-packages/phy/cluster/manual/__init__.py", line 9, in <module>
    from .views import WaveformView, TraceView, FeatureView, CorrelogramView
  File "/Users/lawelie/miniconda3/envs/phy/lib/python3.5/site-packages/phy/cluster/manual/views.py", line 16, in <module>
    from vispy.util.event import Event
ImportError: No module named 'vispy'
(phy) Ingrids-MBP:~ lawelie$ 

> On Oct 23, 2017, at 6:49 AM, Cyrille Rossant <notifications@github.com> wrote:
> 
> python -c "import phy, phycontrib"

rossant, 2017-10-23:
OK, then do:

```bash
pip install vispy requests traitlets tqdm six joblib click
```

and try again?

ingridvanwelie, 2017-10-23:
Ok, so after I did that successfully and try and run physical template-gui params.py again, I now get the following error:

(phy) Ingrids-MBP:~ lawelie$ phy template-gui params.py
Usage: phy template-gui [OPTIONS] PARAMS_PATH

Error: Invalid value for "params-path": Path "params.py" does not exist.

rossant, 2017-10-23:
You need to run this command from a directory where you have the data
Including a params.py file

wrote:
>
> OK, then do:
>
> pip install vispy requests traitlets tqdm six joblib click
> and try again?
https://github.com/kwikteam/phy/issues/775#issuecomment-338622670>, or mute
the thread <https://github.com/
auth/AeYiA0FAZYSAfZSoHuL_xWAoUP18mc9mks5svHDSgaJpZM4QA-FN>.

<https://github.com/kwikteam/phy/issues/775#issuecomment-338638413>, or mute
the thread

ingridvanwelie, 2017-10-23:
Right of course - all working now, thanks for the feedback!

ahudetz, 2018-05-22:
I have the same problem on Mac running Sierra 10.13.4.  I get the message: Error: No such command "template-gui".  Have miniconda3 and Python 3.6 installed.  I did all the suggested steps above but no success.  There was an issue with missing PyQt4 - no source was found for version,  pip was unable to find it.  I tried to install PyQt4 with conda, it seemed to work but still no template-gui.  Please advise.

closed 2019-06-13 by rossant