cortex-lab/phy #777: template-gui fills memory and crashes computer

Kelarion, 2017-12-05:
I'm using phy on Neuropixels data that was processed with kilosort, but when I run template-gui my 64GB of ram gets filled up rapidly, leading to a computer crash if I don't abort phy:

[image]

The strange part is that phy has worked fine on this machine before, and I didn't make any major changes. 

It seems that phy might be loading the entire *.bin file into memory, which seems strange. When I run Phy on a much smaller dataset, it is fine with memory and manages to open the phy window, but then crashes when loading the data. `phy --debug` doesn't return any issues either.

I'm using the latest Anaconda, with python 3.6.3. Maybe this is an issue with my python? I have reinstalled Anaconda and it's my only python environment. (Windows, BTW.)

Any idea what this might be? Many thanks in advance.

NB I'm running from the appropriate directory, and I get the same problems after reinstalling the latest version of phy and phycontrib (`pip install phy phycontrib --upgrade`). This also happens with both the vanilla phy and development version.

Kelarion, 2017-12-05:
This is what I mean by 'crashes when loading data':
[image]

Kelarion, 2017-12-05:
I fixed it by deleting the */.phy folder first created created by:
`[D] config:87            Creating default phy config file at 'C:\Users\Matteo A\.phy\phy_config.py'.`
and then creating a new one (by running template-gui). 

It seems like it was an issue with */.phy/TemplateGui/state.json, since the old and new phy_config files are identical. Beyond that, I can't say.

closed 2017-12-05 by Kelarion

rossant, 2017-12-06:
Good to know! It might have been the trace view config which specified a huge loading interval... We should probably put an upper limit on the window time interval of the trace view.