cortex-lab/phy #778: No TraceView or CorrelogramView, Broken TraceView

coopergrossman, 2017-12-08:
Hello! 

I've been trying to figure this out for awhile now, so maybe someone can help me. Took me awhile just to get the GUI to run; I had to downgrade PyQt and a few of the other python packages to earlier versions. Now it will open, but it looks like the attached picture, where there is no TraceView, Correlogram view and the other windows do not look like the example pictures. Does anyone know what's going on? The other picture shows the packages that I have installed, as I think it might be some compatibility issue. 

[image]
[image]

irisbhy, 2018-02-15:
Hi- I've been having a similar problem, let me know if you ended up finding a solution!

coopergrossman, 2018-02-15:
@irisbhy It ended up being a simple (and silly) fix for me. It was just that the traces were scaled in an absurd way. Use the shortcuts under the TraceView dropdown menu to adjust. Alt+down helped me specifically.

jvschweim, 2018-07-06:
Same here, it happened after I had a memory error and deleted the state.json file. @coopergrossman's tip solved the issue.

wehrlab, 2019-03-06:
Had this again -- in my case the waveform view had ludicrous scaling in the state.json file. Since scaling is not too expensive to compute and depends on the scale of the data (which may be different across datasets) it seems like scaling should be computed fresh on drawing, rather than stored in a global state file -- that or make a state file for each dataset. Can't imagine a case where I would want the GUI state including scaling, etc. preserved across datasets

rossant, 2019-06-13:
@wehrlab good point, done in the dev branch

closed 2019-06-13 by rossant