cortex-lab/phy #780: Number of spikes in feature view

bapungiri, 2018-01-22:
Is it possible to increase/change the number of features shown per cluster ? I would like to have an option to show all spikes in the feature view.

rossant, 2018-01-24:
You should be able to change this by creating a plugin in `~/.phy/phy_config.py`

```python
class MyPlugin(IPlugin):
    def attach_to_controller(self, controller):
        controller.n_spikes_features = 10000
```

see http://phy.readthedocs.io/en/latest/config/

bapungiri, 2018-01-25:
Unfortunately it didn't work. I also tried deleting the .phy located in the 'params.py' folder, but that didn't change the number of spikes either. Is it possible to add this as a menu option in future relases of phy GUI.

rossant, 2019-06-13:
done in the dev branch, as a plugin example

closed 2019-06-13 by rossant