cortex-lab/phy #782: Filter units in the ClusterView according to shank group

petersenpeter, 2018-02-23:
Hi 

I am using phy on data sorted with Kilosort. It would be really helpful if one could only show units belonging to a single shank group. Is there any way one can filter the displayed units in the ClusterView based on shank group? Thanks.

/ Peter

brendonw1, 2018-03-22:
Seconded

jvschweim, 2018-05-09:
It would be great to be able to sort it either to shank groups or in my case separate tetrodes!

petersenpeter, 2018-05-09:
Cyrille, is there any chance this could be become a built in feature in Phy? It would make the workflow much more fluent if a channel group filter/setting was implemented in phy.

rossant, 2018-05-09:
Sure, development has been halted for a long time but it should continue very soon

petersenpeter, 2018-05-16:
Fantastic. Great to hear

petersenpeter, 2018-05-16:
@jvschweim I have made a few plugins for phy that allows you to sort the units by shanks/groups, among other features. It does require you to alter the Kilosort output to include shank information, as this is not part of the standard outputs. You can download the plugins here: http://buzsakilab.com/content/files/PhyPlugins.zip

You can see how to implement it in Kilosort in our wrapper here: https://github.com/brendonw1/KilosortWrapper

@rossant In the plugins above I have also added a resort feature using KlustaKwik, and a noise-artifact removal call that uses the mahalanobis distance to remove outliers with a greater distance than a user-defined threshold. If you like, you are welcome to built them into the standard phy package. Also added firing rate to the units column. 

btw, it would be great if there existed a Phy plugin directory/repository.

jvschweim, 2018-05-16:
Great, I will try them out over the next couple of days @petersenpeter !

rossant, 2019-06-13:
new repo for phy plugins https://github.com/cortex-lab/phy-plugins

petersenpeter, 2019-06-13:
Great with the repo for plugins, but what about the other requests?

rossant, 2019-06-13:
The latest version lets you load arbitrary tsv files with cluster information, like the shank. There is also a new filtering feature in the cluster view that you could use to filter by shank.

If the shank information was added into the KilSort file format, we could integrate this feature natively in phy.

petersenpeter, 2019-06-13:
Great, I have not had time to try out the new version. I added the shank info in the export call in rezToPhy in our Kilosortwrapper. In phy, I created a plugin that adds a column with the shank number, but this only allows us to sort by shank with our current version and not filter: https://github.com/brendonw1/KilosortWrapper/blob/master/rezToPhy_KSW.m
Line 90:
writeNPY(rez.ops.kcoords, fullfile(savepath, 'channel_shanks.npy'));

rossant, 2019-06-13:
Good to know! 
Originally, phy was designed such as to handle only one shank at the time. The reason was that we thought it only made sense to spike sort one shank at a time. Could you elaborate on your use case where you have multiple shanks at once in phy?

petersenpeter, 2019-06-13:
We always used probes with multiple shanks and with the original Kilosort pipeline, only one output is created that includes all shanks. We have therefore worked with multiple shanks in phy. So to improve this I added the shank info as a column in Phy to be able to easier work with the units within shanks. This is not optimal but OK. It would be much better to be able to work with the shanks individually, i.e. filter such that only the units belonging to the selected shank(s) were shown.

rossant, 2019-06-13:
Okay, thanks. So originally, phy was written for klusta, which treated shanks separately. But KiloSort treats all shanks at once. @marius10p Does KiloSort 2 also work by considering all shanks at once? Would you consider modify the export function in KiloSort 2 so as to save the shank number for each cluster ? Then phy could use this information to only show clusters belonging to a given shank.

yger, 2019-06-14:
I agree this would be pretty cool. Currently, for us, I am encouraging people that have shanks to go for a sparse export, such that templates are restricted to electrodes within a shank. This makes things easier for visualization. In fact, in the probe file, you do have everything you need, don't you? Assuming the probe layout is given in in the form of channel_groups[i], where i is the shank, then when you have the best_electrode of a template, you can guess its shank, don't you? And you can even restrict the display of the template/waveforms to the appropriate shank (if phy is still colorcoding arbitrarly, without the sparse export, a fixed number channels near the best_elec)

rossant, 2019-06-14:
The phy file format contains the npy and tsv files, but the prb file is not part of it. It is a relic of the klustakwik era and we don't have it in most kilosort datasets. Currently, phy only uses `channel_positions.npy`. So perhaps something like `channel_shanks.npy`, which would be a `(n_channels,)` array with the shank index of every array, would do it. Phy could load this and I could implement basic multi-shank support

petersenpeter, 2019-06-17:
Exactly. The naming is not important, but with this info, via the peak waveform channel, the shank can be determined in Phy.

yger, 2019-06-17:
Would be perfect for me too!

rossant, 2019-06-17:
Done! The template GUI will automatically load `channel_shanks.npy` if there is one, and add a `shank` column in the cluster view.

closed 2019-06-17 by rossant

petersenpeter, 2019-06-17:
Great! Looking forward to using it.

brendonw1, 2019-06-18:
I’m not sure I understand how this will work.  Will only one shank be
loaded at a time?   Or will shank be listed In the template info columns?

Also if it’s just loading one shank at a time - When Kilosort finds
waveforms on shank A it often finds template-related channels on shank B.
Those channels on shank B are likely irrelevant usually of course but the
template usually includes them nonetheless.  How will that be handled?
Only display the shank channels?

rossant, 2019-06-18:
Phy will load all spikes, templates, and channels stored in the files. If KiloSort saves all shanks into a single, concatenated dataset across all shanks, all of that will be loaded into phy. If there is a `channel_shanks.npy` file, there will be an additional column `shank` with, for every cluster, the shank of the cluster's peak channel. You can filter the list of clusters in the cluster view using this column as you would with any other column.

Regarding your second question. If the dataset was saved in sparse format, it is up to the spike sorting program to decide which channels every template appears in. If the dataset was saved in dense format, then it is up to phy to decide which channels every template appears in. I think KiloSort uses a dense format.

In phy, the selection of "best channels" for any template is, currently, just the N channels closest to the peak channel, where N is a customizable parameter. These channels will all be included on shank A only if the N closest channels belong to shank A, i.e. if the channel positions of shank B are far enough from those in shank A. In other words, when using multiple shanks in the same dataset, you should ensure that `channel_positions.npy` takes into account the positions of the shanks themselves.

yger, 2019-06-18:
I made some tests here, with Spyking circus and the sparse export, and it
looks great. Sparse export is now an option for us, but it will become the
default in the next release I think, as phy is handling it perfectly, and
it is easier to understand what has been done, under the hood, by the spike
sorting software.

I can confirm that without the sparse export, templates will not be
restricted to shanks by phy. You'll only have the extra "shank" column
telling you in which shank the peak is, but template will cover all
channels next to the best one, as Cyrille juste explained

rossant, 2019-06-18:
If that is desirable, I could easily restrict best channels to the current shank.

yger, 2019-06-18:
I think it would make sense, if easy to do for phy. We are not using shanks
ourselves, but people that have shanks far apart want to visualize only the
local information, and not be distracted by putative peaks elsewhere, not
relevant for the local activity next to the best channel, within the shank.

rossant, 2019-06-18:
Okay, done!

brendonw1, 2019-06-18:
This is great.  Thank you

So good having you working on Phy so much Cyrille!