cortex-lab/phy #783: Import errors running phy

hamdanspam, 2018-03-03:
I made sure to follow all directions very carefully, and even purged and reinstalled miniconda before attempting an installation (this is my first time trying to install). When I try to run `phy` from my command line on Windows 7, I get an

> ImportError: No module named traitlets.config

I also included an image of my conda terminal's output.

[image]

rossant, 2018-03-12:
could you do `pip install traitlets` and try again?

hamdanspam, 2018-03-15:
umpteenth reinstall and:
[image]

hamdanspam, 2018-03-15:
It looks like it got installed on python2 instead of python 3. Traitlets are installed on python2, but some parts of the installation got routed to the wrong version of python somehow. What can I do to make sure conda installs on python 3?

rossant, 2018-03-17:
There is a conflict between two installations of Python. Uninstall the one installed in C:\Python27 and try again.

closed 2019-06-13 by rossant