cortex-lab/phy #785: Bug: Sometimes deletes contents of external cluster_NNN.tsv files

weallen, 2018-05-19:
I've been generating TSV files with other quantitative features (SNR, peak to peak amplitude, etc). While Phy seems to load them correctly, sometimes when I am sorting it will suddenly set the values of all externally loaded .tsv files to NaN in the phy GUI. When I then try to open those files in a text editor, everything except for the cluster_id and feature column labels at the top of the file have been deleted. This is pretty annoying because some of these features take a while to compute, and I have to regenerate all the .tsv files for that experiment.

weallen, 2018-05-23:
Attached is an example of the type of file. It seems to happen randomly almost every time during the process of manual merge + good/MUA/noise classification, usually upon either doing a merge or noise classification. I haven't figured out the pattern yet to the point where I could systematically reproduce it.

[cluster_amp.tsv.zip](https://github.com/kwikteam/phy/files/2031990/cluster_amp.tsv.zip)

rossant, 2019-06-13:
should be fixed in the dev branch

closed 2019-06-13 by rossant