cortex-lab/phy #787: Some simple questions about phy template-gui display

khl02007, 2018-05-26:
Hi, not sure if this is the right place to ask these questions, but I could not find the answers from the previously answered issues or the documentation so here they are:

- How exactly is the 'similarity' computed? Is this similarity of the waveform shapes? Distance between clusters in the feature space?
- I assume the waveforms shown in WaveformView are only a small subset of all the spikes that belong to that cluster. How many are being shown? And how many are averaged over when I press 'w'? How is this subset chosen - are they simply the first N waveforms or are they chosen randomly? Is there a way to change the number of waveforms being shown?
- Is the cross correlogram also done with a subset of spikes or is it done with the entire spike train across the whole recording?
- The documentation says: "Template scale factor is template-dependent so the points plotted are not necessarily comparable across templates." Does this mean that when I'm comparing two clusters and one of them has consistently higher template, I should just ignore that? Is there a way to make the amplitudes of the two clusters being compared to be on the same scale?

Thanks so much!

galenlynch, 2018-11-05:
> I assume the waveforms shown in WaveformView are only a small subset of all the spikes that belong to that cluster. How many are being shown? And how many are averaged over when I press 'w'? How is [...]

+1 to this question.. I would love to be able to display all points in the Waveform and Feature views.

czuba, 2018-12-13:
All things that would still be great to know. Thanks in advance for taking the time to answer these

rossant, 2019-06-14:
> * How exactly is the 'similarity' computed? Is this similarity of the waveform shapes? Distance between clusters in the feature space?

@marius10p could you point me where we can find the description of the similarity matrix? Then perhaps I could add it to the phy documentation as well.

> * I assume the waveforms shown in WaveformView are only a small subset of all the spikes that belong to that cluster. How many are being shown? And how many are averaged over when I press 'w'? How [...]

This is a subset of 10x10 waveforms by default: that is, you take 10 uniformly spaced positions in time, and you extract 10 successive spike waveforms at each position (a bit of disk contiguity is necessary for performance reasons). These numbers can be changed in a plugin as explained in the documentation of the development version (`controller.n_spikes_waveforms` and `controller.batch_size_waveforms`). If you have a SSD, you can try increasing these numbers as long as performance is satisfactory. Note that, as a current limitation, you may have to delete the  `.phy` subdirectory in your data directory every time you change these numbers. This is because the extracted waveforms are cached there.

The mean waveforms are computed based on this subset.

> * Is the cross correlogram also done with a subset of spikes or is it done with the entire spike train across the whole recording?

It is also computed with a subset of spikes, `controller.n_spikes_correlograms`, is 100 000 by default. This number corresponds to the maximum number of spikes *for all selected clusters*. If you select only 1 cluster that has less than 100 000 spikes, then all of its spikes are used for the computation of the auto-correlogram.

> * The documentation says: "Template scale factor is template-dependent so the points plotted are not necessarily comparable across templates." Does this mean that when I'm comparing two clusters and [...]

I'm not sure where this bit comes from, but at least in the dev branch, the scale is the same across all selected clusters, so you can safely compare waveforms and templates of different selected clusters. However the scale changes when you switch to another selection of clusters.

Sorry for the delay answering, I'll add some of the answers in the documentation! Please let me know if you need further clarifications.

closed 2019-06-14 by rossant