cortex-lab/phy #790: IndexError for Viewing Clusters

Sarruedi, 2018-06-07:
I run into an index error issue with a dataset that I resorted tweaking the kilosort parameters.
Interestingly the error is not thrown for the first cluster that I select after opeinng the sorted data with phy but essentially for any cluster after the first one I get the error below and the display does not update. The opening of the file does not lead to the generation of the .phy folder in the directory which contains the data, which I suspect might be the cause of the error but I'm not sure what to change at this point. 
Does anybody have experience with this kind of error?
Here's an example of the error message:

2018-06-07 09:21:08 [D] gui_component:393    Select clusters: 1.
2018-06-07 09:21:11 [D] gui_component:393    Select clusters: 2.
2018-06-07 09:21:11 [E] cli:39               An error has occurred (IndexError): index 3374910 is out of bounds for axis 1 with size 3371858
2018-06-07 09:21:11 [D] cli:42               Traceback (most recent call last):
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\gui\widgets.py", line 196, in _emit_from_js
    self.emit(text_type(name), json.loads(text_type(arg_json)))
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\gui\widgets.py", line 115, in emit
    return self._event.emit(*args, **kwargs)
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\utils\event.py", line 121, in emit
    res.append(callback(*args, **kwargs))
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\cluster\manual\gui_component.py", line 334, in on_select
    self._emit_select(cluster_ids)
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\cluster\manual\gui_component.py", line 395, in _emit_select
    self.gui.emit('select', cluster_ids)
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\gui\gui.py", line 228, in emit
    return self._event.emit(*args, **kwargs)
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\utils\event.py", line 121, in emit
    res.append(callback(*args, **kwargs))
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\cluster\manual\views.py", line 1220, in on_select
    x = self._get_feature(x_dim[i, j], spike_ids, f)
  File "c:\users\srl_kilosort\miniconda3\envs\phy\lib\site-packages\phy\cluster\manual\views.py", line 1056, in _get_feature
    return -1. + (2. / self.duration) * self.spike_times[spike_ids]
IndexError: index 3374910 is out of bounds for axis 1 with size 3371858

Sarruedi, 2018-06-07:
I just reinstalled phy (upgraded to DEVELOPER VERSION) and now I can open the same data. 
So it's seems the sorting itself is fine and it's about a error within the non developer version of phy.

galenlynch, 2018-08-29:
I also get this behavior.

closed 2019-06-13 by rossant