cortex-lab/phy #793: Undocucmented developer version features

stephengv, 2018-06-27:
I recently switched over to the developer version of Phy, and have a few questions about features that I don't believe were present in our previous version. I am using Kilosort with the Phy Template-GUI. 

1) In Feature View, we now see grey dots that weren't showing up in our previous version. Do these points represent other spikes detected on that channel that are not a part of the selected cluster? (presumably from other clusters and noise?)

if so, are they every other detected spike, or a subset? If a subset, is it randomly selected, and can we control how many to display?

[image]

2) In Amplitude view, we now see a grey line that wasn't there previously. Does this grey line represent the threshold value used by kilosort to initially detect spikes?

[image]

3) And finally, when we select a cluster, the development version takes 3-5 seconds to update the display and switch to that cluster. the stable release of Phy we were using previously seemed to do that switch immediately on the same hardware. Are their any specific parameters we could adjust to speed up this process?

Thanks so much! we are really starting to get the hang of Phy, and are impressed by what it is capable of!

PhantomSpike, 2018-08-28:
Hi,

I have also switched to the developer version and pretty much have the same questions!

Thank you.

rossant, 2019-06-14:
> 1. In Feature View, we now see grey dots that weren't showing up in our previous version. Do these points represent other spikes detected on that channel that are not a part of the selected cluster? (presumably from other clusters and noise?)

Gray dots represent background features. There are 1000 by default, you can change this number with `controller.n_spikes_features_background` in a plugin, as explained in the documentation of the development version.

See also https://github.com/cortex-lab/phy/issues/787#issuecomment-501984450

> 1. In Amplitude view, we now see a grey line that wasn't there previously. Does this grey line represent the threshold value used by kilosort to initially detect spikes?

This was a line showing y=1, but it is gone in the latest development version, because there are proper axes now.

> 1. And finally, when we select a cluster, the development version takes 3-5 seconds to update the display and switch to that cluster. the stable release of Phy we were using previously seemed to do that switch immediately on the same hardware. Are their any specific parameters we could adjust to speed up this process?

This should be fixed in the latest version.

closed 2019-06-14 by rossant