cortex-lab/phy #794: Amplitude Histogram Plugin

stephengv, 2018-06-27:
@lshaheen

We have recently installed the [Amplitude Histogram](https://github.com/kwikteam/phy-contrib/blob/master/phycontrib/views/amplitude_histogram.py) plug in for phy (thanks!), and have a few questions.

1) We are reciving lots of error reports via the cmd window while running phy with the plugin that we don't understand. Does anyone have any insight on these? 

[image]

2) I recall reading that this plugin would also estimate the proportion of "missing" spikes based on the assumption of normality. Is this true? and if so, where does this output appear?

3) Finally, in the image below, what does the dotted grey line represent?

[image]

Sorry if this is the wrong place to post this, please let me know if there is somewhere more appropriate to post this question.

lshaheen, 2018-06-28:
Hi,
Glad this is useful. 
1: The "Fitting failed with..." errors are saying that the function wasn't able to fit the histogram and is trying again with more iterations. The the "Fitting error: Optimal..." means it tried again and still failed. These are cases where the amplitude histogram doesn't look like a gaussian at all. Usually these are not single-units, or there was a lot of drift in the recording.

2: Yes. In a more recent version I printed the number on the graph, you can get that at https://github.com/kwikteam/phy-contrib/pull/105  Otherwise if you keep the version you have now and open phy with --debug at the end the numbers will print on the console.

3: The dashed line represents what the gaussian "should" like if there weren't missing spikes.

For example, here's one that shouldn't be called a single-unit, because even though the autocorrelogram shows very few refractory violations, the amplitude histogram shows the cluster is missing a bunch of spikes.
[image]

Here's a good single-unit where the histogram isn't cut off.
[image]

But beware that if there's drift the amplitude histogram often doesn't look like a single gaussian, so the fit is poor and that number doesn't mean anything. In this example, there are two peaks so the fit is poor, but it doesn't look cut off at all so it's still a single-unit.
[image]

In the case you showed the histogram looks really cut off so I would say it's more than likely not a single-unit.

stephengv, 2018-06-29:
Thanks so much! we really appreciate the examples.

closed 2019-06-13 by rossant