cortex-lab/phy #797: Show more PC's in Feature View?

lauritk, 2018-08-13:
I tried to modify the code so, that I could see more principal components in the feature view. I modified the 'feature.py' to have component C, and changed 'n_cols' to 6. I get error msg:

'10:46:03 [E] cli:36               An error has occurred (ValueError): too many values to unpack (expected 2)'

Any ideas, where the error comes from?

rossant, 2018-08-13:
no, please feel free to post your code and put a more detailed error message by using the `--debug` option to phy

lauritk, 2018-08-14:
Of course, sorry.

Got rid of the previous error. Changed tabs I used in the 's' string to spaces. Now got different error tho.

> 09:28:50 [D] supervisor:342       Select cluster(s): 45.
> 09:28:50 [E] cli:36               An error has occurred (ValueError): invalid literal for int() with base 10: 'A'
> 09:28:50 [D] cli:39               Traceback (most recent call last):
>   File "c:\devel\anaconda3\lib\site-packages\phy\cluster\views\base.py", line 92, in update_view
>     self.on_select(cluster_ids, **kwargs)
>   File "c:\devel\anaconda3\lib\site-packages\phy\cluster\views\feature.py", line 293, in on_select
>     self._plot_points(i, j, dim_x, dim_y, background)
>   File "c:\devel\anaconda3\lib\site-packages\phy\cluster\views\feature.py", line 185, in _plot_points
>     py = self._get_axis_data(bunch, dim_y, cluster_id=cluster_id)
>   File "c:\devel\anaconda3\lib\site-packages\phy\cluster\views\feature.py", line 155, in _get_axis_data
>     c_rel = int(dim[:-1])
> ValueError: invalid literal for int() with base 10: 'A'

And the code is:

>          ...
>          def _get_default_grid():
>              """In the grid specification, 0 corresponds to the best channel, 1
>              to the second best, and so on. A, B, C refer to the PC components."""
>              s = """
>              time,0A 1A,0A   0B,0A   1B,0A   0C,A0   1C,A0
>              0A,1A   time,1A 0B,1A   1B,1A   0C,1A   1C,1A
>              0A,0B   1A,0B   time,0B 1B,0B   0C,0B   1C,0B
>              0A,1B   1A,1B   0B,1B   time,1B 0C,1B   1C,1B
> 	         0A,0C   1A,0C   0B,0C   1B,0C   time,0C 1C,0C
> 	         0A,1C   1A,1C   0B,1C   1B,1C   0C,1C   time,1C
>              """.strip()
>              dims = [[_ for _ in re.split(' +', line.strip())]
>                      for line in s.splitlines()]
>              return dims
>          ...
>          assert features
>          self.features = features
> 
>          self.n_cols = 6
>          self.shape = (self.n_cols, self.n_cols)
> 
>          self.grid_dim = _get_default_grid()  # [i][j] = '..,..'
>          ...

lauritk, 2018-08-14:
> first line, last column, should be 0A and not A0?

Oh, I must be blind! Thanks, it is working now.

closed 2018-08-14 by lauritk