cortex-lab/phy #798: No module named 'PyQt4' even after installing

DJ-Hayden, 2018-08-20:
Hello, 
I'm trying to get to work on some updates using the development version, but it seems like I cannot get PyQt4 on my Ubuntu 16.04 system.

I do have it installed:

'''python3-pyqt4 is already the newest version (4.11.4+dfsg-1build4)'''

yet the output of:

'''
python -c "import phy.gui.qt"
'''

is:

'''
  File "<string>", line 1, in <module>
  File "/home/dustin/miniconda3/lib/python3.6/site-packages/phy/gui/__init__.py", line 6, in <module>
    from .qt import require_qt, create_app, run_app
  File "/home/dustin/miniconda3/lib/python3.6/site-packages/phy/gui/qt.py", line 21, in <module>
    from PyQt4.QtCore import (Qt, QByteArray, QMetaObject, QObject,  # noqa
ModuleNotFoundError: No module named 'PyQt4'
'''

I've scoured the Internet and can't seem to find anything about why this is happening. Any suggestions?

DJ-Hayden, 2018-08-20:
Update:

I solved this as follows:
1) Download the provided environment.yml file
2) run ```conda env create --name phy```
3) run ```source activate phy```
4) run ```conda install pip git``` (to have the developer version be completely separate from other code)
5) run ```pip install git+https://github.com/kwikteam/phy git+https://github.com/kwikteam/phy-contrib --upgrade``` (to have the developer version be completely separate from other code)
6) rerun master_eMouse in MATLAB (for safety, didn't know if there'd be some weird things happening with old py files from the broken version)
7) cd to your eMouse Fake Data
8) run ```phy template-gui params.py```

This works great (at least for the master_eMouse script, I haven't tried the singlePhase3 or dualPhase3 scripts. 

However, there is still one consistent warning (that I've received even in the stable release):
```
WARNING: Qt: Cannot set locale modifiers: 
14:25:36 [W] _qt:160              Qt: Cannot set locale modifiers: 
```
Since I imagine it is still a PyQt4 option, I figured I wouldn't open another bug report and instead just lump it into this one.

rossant, 2019-06-13:
fixed in the dev branch with PyQt5 update

closed 2019-06-13 by rossant