cortex-lab/phy #799: Assertion error in cluster_manual

galenlynch, 2018-08-29:
This set of files was working fine yesterday, though I often got indexing bugs when try to view one or two clusters. Now I can't open the clustering resuts, due to the following error:
```
2018-08-29 13:36:14 [D] gui:253              Loading traces at `flatfile`.
2018-08-29 13:36:14 [D] gui:268              Loading amplitudes.
2018-08-29 13:36:14 [D] gui:277              Loading 155547 spike clusters.
2018-08-29 13:36:14 [D] gui:283              Loading spike templates.
2018-08-29 13:36:14 [D] gui:289              Loading spike samples.
2018-08-29 13:36:14 [D] gui:293              Loading templates.
2018-08-29 13:36:14 [D] gui:299              Loading the whitening matrix.
2018-08-29 13:36:14 [D] gui:303              Loading unwhitened templates.
2018-08-29 13:36:14 [D] gui:321              Loading similar templates.
2018-08-29 13:36:14 [D] gui:326              Loading channel mapping.
2018-08-29 13:36:14 [D] gui:334              Loading channel positions.
2018-08-29 13:36:14 [D] gui:339              Loading features.
2018-08-29 13:36:14 [D] gui:370              Loading template features.
2018-08-29 13:36:14 [D] gui:424              HP filtering the data for waveforms
2018-08-29 13:36:14 [D] gui:453              Loading the cluster groups.
2018-08-29 13:36:14 [D] context:56           Initialize joblib cache dir at `/home/glynch/mnt/feevault/Analysis/kilosort/results/192243/testing/.phy`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_masks`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_features`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_waveforms`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phycontrib.template.gui.get_waveforms_amplitude`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phycontrib.template.gui.get_waveform_lims`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_feature_lim`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_trace_lim`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_close_clusters`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_probe_depth`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_correlograms`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phy.cluster.manual.controller.spikes_per_cluster`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phycontrib.template.gui.get_amplitudes`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phycontrib.template.gui.get_cluster_templates`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phycontrib.template.gui.get_cluster_pair_features`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phycontrib.template.gui._cluster_template_similarities`.
2018-08-29 13:36:14 [D] context:80           Load memcache for `phycontrib.template.gui._sim_ij`.
2018-08-29 13:36:15 [E] cli:39               An error has occurred (AssertionError): 
2018-08-29 13:36:15 [D] cli:42               Traceback (most recent call last):
  File "/home/glynch/anaconda3/envs/phy/bin/phy", line 11, in <module>
    sys.exit(phy())
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phycontrib/template/gui.py", line 814, in cluster_manual
    _run_cmd('_run(params)', ctx, globals(), locals())
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phy/utils/cli.py", line 76, in _run_cmd
    return exec_(cmd, glob, loc)
  File "<string>", line 1, in <module>
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phycontrib/template/gui.py", line 786, in _run
    controller = TemplateController(**params)
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phycontrib/template/gui.py", line 249, in __init__
    super(TemplateController, self).__init__()
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phy/cluster/manual/controller.py", line 70, in __init__
    self._set_manual_clustering()
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phycontrib/template/gui.py", line 458, in _set_manual_clustering
    super(TemplateController, self)._set_manual_clustering()
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phy/cluster/manual/controller.py", line 161, in _set_manual_clustering
    new_cluster_id=new_cluster_id,
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phy/cluster/manual/gui_component.py", line 161, in __init__
    new_cluster_id=new_cluster_id)
  File "/home/glynch/anaconda3/envs/phy/lib/python3.5/site-packages/phy/cluster/manual/clustering.py", line 161, in __init__
    assert np.all(self._spike_clusters < self._new_cluster_id)
AssertionError

```

rossant, 2018-08-29:
Could you rename the `.phy`  subdirectory (within the data folder) into `.phy~` and try again? The cache might be corrupted somehow, with this manipulation it will be recreated

galenlynch, 2018-08-29:
I fixed the problem by updating to the development version, and deleting the `.phy` directory. I had already deleted `.phy` on the regular version, which did not solve the problem.

Thanks for the quick response!

closed 2018-08-29 by galenlynch