cortex-lab/phy #800: Problem with TraceView and WaveformView

kevin-allen, 2018-09-03:
First, thanks for sharing phy.

I am trying to set up KiloSort and phy on an Ubuntu 18.04 installation. KiloSort runs fine, as far as I can tell. 

I am having problems with the TraceView and WaveformView in phy. See the pictures below.
[image]

[image]

Some data points go out of scale.

My first suspicion was that something was wrong with the .dat file, but when I open it with neuroscope or other software (my relectro R package) it looks normal (see below).

[image]

I have uploaded all configuration files, master file and the .dat file to my google drive if someone wants to try to replicate this problem.
https://drive.google.com/open?id=1ExiikRxrtQBPP0AdPibtG5qadUbvkirU

Any ideas what could be causing the traces to go out of the normal range?

Many thanks for any potential feedback.

rossant, 2018-09-05:
It might be a fitting problem related to kilosort? If I'm not mistaken, the large vertical lines you see are the residuals, ie the traces minus the templates. Does this happen on other datasets?

Sergudx, 2018-11-07:
I'm experiencing the same in some of the Kampff Lab recordings (not in all of them). I'm having the problem just when using Kilosort for the clustering (not with Klusta, for example).

Any thoughts on what's going on?

rossant, 2019-06-13:
please re-open if the issue persists in the dev branch of phy

closed 2019-06-13 by rossant