cortex-lab/phy #808: Calculating raw Amplitude

alex-attinger, 2018-10-19:
Just starting to use phy with template gui extension. Thanks a lot for this great tool! I would like to have an estimate of the spike amplitude on its best channel. Currently, I am doing this via a plugin, where I first get spikeIDS (spike_ids = controller._select_spikes(cluster_id, 500, batch_size=None), then the waveforms (wf=controller.all_waveforms[spike_ids]). Those I then average and take the min and max to get an estimate. However, this for some reason slows down phy startup dramatically (several minutes vs a few seconds). Any hints in how I can make this faster?

rossant, 2018-10-21:
I think using [`_get_waveforms()`](https://github.com/kwikteam/phy-contrib/blob/master/phycontrib/template/gui.py#L233) could be faster.

alex-attinger, 2018-10-22:
Thank you for your reply. In my version of phycontrip (1.0.15), installed with pip, the function is still called get_waveforms(), as far as I can tell. There, the waveforms are normalized before being returned. Should I update to the latest version? How would I do that (I already ran pip install phy phycontrib --upgrade ).

alex-attinger, 2018-10-23:
So I've updated to the latest version. However, the simple plugin structure for adding columns to the clusterview no longer works, as the variable "manual_cluster" in the templatecontroller class no longer exists. I can of course edit the source code directly, but I liked the idea of a plugin that is separate from the core code much better. Any suggestions?

rossant, 2018-10-24:
this object was renamed to `controller.supervisor` so the plugin should work if you use the new name
here is an example of a plugin: https://github.com/kwikteam/phy-contrib/blob/master/phycontrib/views/amplitude_histogram.py

alex-attinger, 2018-10-26:
Hi, I tried renaming it to controller.supervisor. This generates a different error (An error occurred when attaching plugin CorrPlugin: 'Supervisor' object has no attribute 'cluster_view'.) Not sure why that is happening, maybe because at the time the plugin is attached, the supervisor does not yet have a cluster_view? Anyway, I got the measures I wanted by directly editing the gyi.py file, so it is not urgent for me.

closed 2019-04-30 by alex-attinger