cortex-lab/phy #809: Waveform channel mapping

johnleeuk12, 2018-12-06:
Hi,

Thank you for developping Phy!
I'm currently using template-gui with kilosort, and cannot seem to get more than one row of channel waveforms with Template-gui. I'm using the eMouse code from Kilosort, and it seems to be displaying channels with a unique y coordinate (regardless of x). Overlapping y coordinates are not shown in the gui, and x coordinates does not seem to have an effect. I've verified this with the .npy Channel position and Channel map files.  Has anyone else had a similar issue?

johnleeuk12, 2018-12-06:
I would like to also add that I'm using phy 1.0.9, and when I look at Probeview, both columns are there, but Waveform view only shows me one. This is a screenshot of the test dataset from
 https://phy-contrib.readthedocs.io/en/latest/template-gui/#test-datasets

[image]

rossant, 2018-12-17:
maybe try to delete `~/.phy/` to reset the view state of the waveform view?

johnleeuk12, 2019-01-11:
Yep. That worked. Thank you!

closed 2019-01-11 by johnleeuk12